ARMmbed/mbed-os #2059: main thread task_id is 1 when mbed is configured with #define OS_TIMERS 0

gdhamp, 2016-06-28:
When mbed is built with #define OS_TIMERS 0, svcThreadCreate() fails (correctly) for the creation of the timer thread. This causes the main thread to have task_id 1. There are a couple of places where the task_id for the main thread is assumed to be 0x02. In particular, rt_stk_check fails since it is trying to ignore the MAGIC_WORD check on the main thread but the task_id check fails since the main thread task_id is 0x01 rather than 0x02.

0xc0170, 2016-06-29:
Thanks for the reporting, I'll send a patch to fix this

0xc0170, 2016-06-30:
I can't reproduce the locally this issue (using small build for gcc arm), please provide more details.

I'll try to manually disable timers and run gdb. I'll report back

0xc0170, 2016-06-30:
I was able to reproduce it, at least to check thread id that it's 0x01 for the main thread. The patch is proposed here :https://github.com/mbedmicro/mbed/pull/2075.
Please review

PR #2075 (2016-06-30): RTOS - fix for main thread id might not be 0x02
Fixes #2059. As reported, if timer thread is not created, the main thread
id is 0x01. We introduce MAIN_THREAD_ID macro to define the id. We shall consider,
if we keep this in a variable.

I placed MAIN_THREAD_ID in cmsis_os.h as that header is safe to include within RTX, not like
RTX_Config.h or RTX_CM_Lib.h). I reported an issue to cmsis team about this inclusion mess within RTX.

tested with K64F with default config:

```
+--------+------------+-----------+---------+-------------------------+--------------------+---------------+-------+
| Resu [... cut]
commit 757d340b (2016-06-30): RTOS - fix for main thread id might not be 0x02

Fixes #2059. As reported, if timer thread is not created, the main thread
id is 0x01. We introduce MAIN_THREAD_ID macro to define the id. We shall consider,
if we keep this in a variable.

I placed MAIN_THREAD_ID in cmsis_os.h as that header is safe to include within RTX, not like
RTX_Config.h or RTX_CM_Lib.h).
closed 2016-06-30 by sg-