ARMmbed/mbed-os #2061: mbedgt dependency not installed

mlnx, 2016-06-28:
Bug on installing dependencies. It affects the user experience at installing mbed tools.

Just tried the mbed tools on a new environment and the cmd tools were not able to find `mbedgt`
Also tried: ```sudo pip install -r requirements.txt``` but the problem persists.

Environment:
Linux Ubuntu 14.04.01
mbed-cli version 0.7.17
mbed #ec1ec7c3e2f383b7d795856756abf6b934f37fd

Workaround:
sudo pip install mbed-greentea --upgrade

bridadan, 2016-06-28:
@mlnx Do you have the output from the initial `sudo pip install -r requirements.txt` command? I can confirm that `mbed-greentea` is listed in `requirements.txt` at the hash you listed: https://github.com/mbedmicro/mbed/blob/ec1ec7c3e2f383b7d795856756abf6b934f37fdf/requirements.txt

So it should have been installed. Unless you had a very old installation of mbed-greentea that didn't have the `mbedgt` alias? We don't currently enforce a version of greentea in the `requirements.txt`, so I'm curious if this was your issue.

bridadan, 2016-06-28:
The requirements should probably be similar to what you proposed here: https://github.com/ARMmbed/mbed-os/pull/386

mlnx, 2016-06-29:
> Do you have the output from the initial pip install -r requirements.txt command? 

There was no output. Indeed `mbed-greentea` was listed as requirement (without version info)

> Unless you had a very old installation of mbed-greentea that didn't have the mbedgt alias?

It's possible I had an old version. And I think quite a few people may hit this case.

> The requirements should probably be similar to what you proposed here

Interestingly, I manually updated my local `requirements.txt` with the suggested versions and then ran `pip install -r requirements.txt` but it didn't fix the issue.  This one did: `sudo pip install mbed-greentea --upgrade`.

I think we need a new clean environment to test this behaviour again.

bridadan, 2016-06-29:
Was it the `sudo` or the `--upgrade` that made it go through? I could see that the `sudo` on the Linux machine may be required. Maybe we need to update the instructions to reflect this?

mlnx, 2016-07-01:
Looking again into the history of commands, I confirm I was using `sudo`, so that's not the problem.

Let's keep this ticket open. I'm not the only one that reported a weird behaviour at installing greentea.

BlackstoneEngineering, 2016-07-01:
Ubuntu 14.4 LTS has a super old version of pip, this is a known issue with packages like setuptools not installing correctly unless you upgrade pip. Please run `pip --version` to see what version you have, should be on at least v8.x.x

mlnx, 2016-07-01:
That's true, my pip was quite old (v6.1) and I've updated it.
I've added a note into the ERP notes.

BlackstoneEngineering, 2016-07-01:
This is an implicit of the requirement to have python 2.7.11+, this implicitly includes a new version of pip as well in the windows installer, should probably be made explicit that you need a newish version of both python 2.7 and pip for stuff to work. This is true for most all of our tools. 

PS: Ubuntu 14.4 LTS has a bunch of known issues, mostly relating to the Ubuntu packages being seriously out of date, like really really far out of date, like I dont use ubuntu 14.4 'cause its almost guaranteed to not work correctly out of the box kind of out of date.

mlnx, 2016-07-06:
I'm closing this ticket for now, as I considered it solved. May re-open in the future if I continue to see problems.

closed 2016-07-26 by MarceloSalazar