ARMmbed/mbed-os #2072: Should be able to define target_overrides in libraries

janjongboom, 2016-06-30:
I have a library that depends on mbed Client. In the past I'd just reference mbed Client as a dependency and be done with it. Now I need to declare the CLIENT feature in mbed_app.json. I added an mbed_app.json file to my library but it's not being picked up by mbed-cli.

That's highly annoying because now I need to tell the user to not only reference my app but also add stuff to this config file, where in the past this would go automatically when he includes my library.

@sg- @screamerbg

bridadan, 2016-06-30:
I think you need to use `mbed_lib.json` instead?

It looks like the config system docs are still in a PR: https://github.com/mbedmicro/mbed/pull/1984/files

janjongboom, 2016-07-01:
Yep, that worked. Thanks.

closed 2016-07-01 by janjongboom