ARMmbed/mbed-os #2081: Build tools may remove entire mbed repo

theotherjimmy, 2016-06-30:
I know this is not a normal invocation. I found this when fuzzing the command line arguments for one of my PR's:
```
$ python tools/make.py -t gcc_arm -m k64f --source . --build . -p 0 --clean
Building project mbed to faoaou (K64F, GCC_ARM)
[ERROR] [Errno 22] Invalid argument: '.'
$ ls
$ git status
fatal: Not a git repository (or any of the parent directories): .git
```
No more mbed repo.
We might want to add a check that prevents this.

0xc0170, 2016-06-30:
A note for this type of bug: The tools should have tests, something to consider.

We should fix this asap

bridadan, 2016-06-30:
They absolutely should!

+1

theotherjimmy, 2016-07-08:
The rule we have to enforce is this: the build directory may not be a parent of the PWD or the mbed repo.

PR #2142 (2016-07-22): [tools-build,make,test] Disallow parent of repo as --build args
Resolves #2081 by making any parent of the repo (or the repo dir itself)
passed to --build fail at argument parsing time.
commit 52855b53 (2016-07-22): [tools-build,make,project] Disallow parent of repo as --build args

Resolves #2081
closed 2016-07-22 by sg-