ARMmbed/mbed-os #2091: TCP socket does not read beyond MSS (Max Segment Size) window.

mottigondabi, 2016-07-04:
### Description
- Type: Bug
- Priority: Major

## Bug
TCP socket does not read beyond MSS (Max Segment Size) window.
Sending a big chunk of data e.g. 3KB from a remote host to the target mBED device that exceeds MSS results only in 1460 bytes read which is probably MSS (Max Segment Size).

Happens over TCP, blocking mode with no timeout set.

**mbed-os sha**
(`b4bb088876cb`)

**mbed-cli version**
(`mbed 0.7.13`)

**Expected Behavior**
The `recv` method should read the entire 3KB

**Actual Behavior**
The `recv` method read only 1460B

0xc0170, 2016-07-06:
cc @geky @bogdanm

bogdanm, 2016-07-11:
In Linux at least, it's perfectly normal for send()/recv() to transfer a number of bytes which is lower than the number of bytes requested by the caller. The caller must check the return value of send()/recv() and keep on going until the whole buffer is sent or received. This is a common "gotcha" of network apps: assuming that send() or recv() will always transfer the requested number of bytes.

bogdanm, 2016-07-12:
@mottigondabi, does this make sense to you?

jenia81, 2016-07-14:
@bogdanm you right about Linux behavior.
I suggest maybe to add documentation about this in your APIs

tommikas, 2016-08-09:
>In Linux at least, it's perfectly normal for send()/recv() to transfer a number of bytes which is lower than the number of bytes requested by the caller.

In this case though, while recv() doesn't transfer more than 1460 bytes, send() happily sends more than that. This small inconsistency makes it that little bit more confusing.

>I suggest maybe to add documentation about this in your APIs

Second that. The behaviour itself is fine, as long as it's clearly documented.

(For example the [mbed OS 2.0 socket API defined both send/recv and sendall/recvall](https://developer.mbed.org/users/mbed_official/code/Socket/docs/434906b5b977//classTCPSocketConnection.html). No confusion possible there.)

bogdanm, 2016-08-09:
> In this case though, while recv() doesn't transfer more than 1460 bytes, send() happily sends more than that. This small inconsistency makes it that little bit more confusing.

I don't think they're supposed to be able to send or receive the exact same number of bytes in a single call. The only thing that doeesn't change is: call send/recv with the size of your buffer, verify the return value, keep on sending/receiving your buffer until you're done.

tommikas, 2016-08-09:
Sure, like I said the behaviour itself is fine, it's just the documentation that could do with some fine-tuning. Otherwise both me and @jenia81 (and others later if not yet) wouldn't have ended up wondering about the same thing.

(By the way, the documentation for both calls is so similar that the description of recv states that "recv blocks until data is _sent_". A copy-paste error perhaps?)

closed 2017-10-27 by ChiefBureaucraticOfficer