ARMmbed/mbed-os #2108: LPC2368 - ARM failure - critical.c not compatible with ARM7 core

0xc0170, 2016-07-06:
The mbed lib build output:

```
python tools\build.py -m LPC2368 -t ARM

Compile: critical.c
[Warning] critical.h@50:  #1295-D: Deprecated declaration core_util_critical_section_enter - give arg types
[Warning] critical.h@63:  #1295-D: Deprecated declaration core_util_critical_section_exit - give arg types
[Warning] critical.c@37:  #223-D: function "__get_PRIMASK" declared implicitly
[Error] critical.c@86:  #20: identifier "tmp_IntEnable" is undefined
[Warning] critical.c@95:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@98:  #223-D: function "__CLREX" declared implicitly
[Warning] critical.c@102:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@107:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@110:  #223-D: function "__CLREX" declared implicitly
[Warning] critical.c@114:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@120:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@123:  #223-D: function "__CLREX" declared implicitly
[Warning] critical.c@127:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@134:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@135:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@143:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@144:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@152:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@153:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@162:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@163:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@171:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@172:  #223-D: function "__strex" declared implicitly
[Warning] critical.c@180:  #223-D: function "__ldrex" declared implicitly
[Warning] critical.c@181:  #223-D: function "__strex" declared implicitly
"mbed/.build/mbed/critical.h", line 50: Warning:  #1295-D: Deprecated declaration core_util_critical_section_enter - give arg types
"mbed/.build/mbed/critical.h", line 63: Warning:  #1295-D: Deprecated declaration core_util_critical_section_exit - give arg types
"mbed\hal\common\critical.c", line 37: Warning:  #223-D: function "__get_PRIMASK" declared implicitly
"mbed\hal\common\critical.c", line 86: Error:  #20: identifier "tmp_IntEnable" is undefined
"mbed\hal\common\critical.c", line 95: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 98: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 102: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 107: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 110: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 114: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 120: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 123: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 127: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 134: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 135: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 143: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 144: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 152: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 153: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 162: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 163: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 171: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 172: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 180: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 181: Warning:  #223-D: function "__strex" declared implicitly
mbed\hal\common\critical.c: 24 warnings, 1 error
"mbed/.build/mbed/critical.h", line 50: Warning:  #1295-D: Deprecated declaration core_util_critical_section_enter - give arg types
"mbed/.build/mbed/critical.h", line 63: Warning:  #1295-D: Deprecated declaration core_util_critical_section_exit - give arg types
"mbed\hal\common\critical.c", line 37: Warning:  #223-D: function "__get_PRIMASK" declared implicitly
"mbed\hal\common\critical.c", line 86: Error:  #20: identifier "tmp_IntEnable" is undefined
"mbed\hal\common\critical.c", line 95: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 98: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 102: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 107: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 110: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 114: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 120: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 123: Warning:  #223-D: function "__CLREX" declared implicitly
"mbed\hal\common\critical.c", line 127: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 134: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 135: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 143: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 144: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 152: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 153: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 162: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 163: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 171: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 172: Warning:  #223-D: function "__strex" declared implicitly
"mbed\hal\common\critical.c", line 180: Warning:  #223-D: function "__ldrex" declared implicitly
"mbed\hal\common\critical.c", line 181: Warning:  #223-D: function "__strex" declared implicitly
mbed\hal\common\critical.c: 24 warnings, 1 error

Completed in: (7.00)s

Build failures:
  * ARM::LPC2368

```

cc @c1728p9 @adbridge

0xc0170, 2016-07-06:
I am looking at this further to fix this. LPC2xxx family (I found at least 2) have VIC peripheral: here's some description of those registers.

What is important for us, it's in the cm7 header file for LPC2368:
```
#define __disable_irq()         unsigned tmp_IntEnable = LPC_VIC->IntEnable; \
                                LPC_VIC->IntEnClr = 0xffffffff

#define __enable_irq()          LPC_VIC->IntEnable = tmp_IntEnable
```

It's obvious that ``tmp_IntEnable`` is defined for disable_irq(), therefore ``__enable_irq`` needs t o be in the same scope,  this is not valid for our critical enter/exit.  The macros look weird even for this target:

```
#define __disable_irq()         unsigned tmp_IntEnable = LPC_VIC->IntEnable; \
                                LPC_VIC->IntEnClr = 0xffffffff

#define __enable_irq()          LPC_VIC->IntEnable = tmp_IntEnable

#elif (defined (__ICCARM__)) /*------------------ ICC Compiler -------------------*/

#define __enable_irq                              __enable_interrupt        /*!< global Interrupt enable */
#define __disable_irq                             __disable_interrupt       /*!< global Interrupt disable */
#define __NOP                                     __no_operation()          /*!< no operation intrinsic in IAR Compiler */ 

#elif (defined (__GNUC__)) /*------------------ GNU Compiler ---------------------*/

static __INLINE void __enable_irq() {
    unsigned long temp;
    __asm__ __volatile__("mrs %0, cpsr\n"
                         "bic %0, %0, #0x80\n"
                         "msr cpsr_c, %0"
                         : "=r" (temp)
                         :
                         : "memory");
}

static __INLINE void __disable_irq() {
    unsigned long old,temp;
    __asm__ __volatile__("mrs %0, cpsr\n"
                         "orr %1, %0, #0xc0\n"
                         "msr cpsr_c, %1"
                         : "=r" (old), "=r" (temp)
                         :
                         : "memory");
    // return (old & 0x80) == 0;
}
```

3 different toolchains - 3 different enable/disable irq :sob: 

I am going to send a patch to disable ARM toolchain until this is fixed.

sg-, 2016-07-08:
I think we should deprecate support for ARM7 and focus only on Cortex

closed 2017-01-16 by sg-