ARMmbed/mbed-os #2110: Hardware flow control support for Serial API

hasnainvirk, 2016-07-06:
### Description
- Priority: Blocker

SLIP driver uses hardware flow control. mbed OS 3 HAL had hooks for that in the Target HAL for FRDM-K64F. For example:

https://github.com/ARMmbed/mbed-hal-ksdk-mcu/commit/9bfcfd0572d1835a889779515a3b612262785a46

https://github.com/ARMmbed/mbed-hal-frdm-k64f/commit/77042cc9453c91957b04e9bd200a2dd928723e5a

For morpheous integration, such changes in HAL are needed. Please look into this.

@0xc0170 @screamerbg @SeppoTakalo

0xc0170, 2016-07-06:
It shall be compatible (the mbed HAL API its the same for this feature). The mbed 3 version uses KSDK1.0, and we have here 2.0 version, that API for UART might have changed.

@hasnainvirk Can you backport it here?

sg-, 2016-07-08:
@bogdanm

bogdanm, 2016-07-11:
Looking at https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_K64F/drivers/fsl_uart.h, I can't see any functions related to the flow control settings. @0xc0170, do you know anything about this?

bogdanm, 2016-07-11:
https://community.nxp.com/thread/352310 :disappointed:

bogdanm, 2016-07-11:
@hasnainvirk, can you check if https://github.com/mbedmicro/mbed/pull/2140 fixes your problems?

hasnainvirk, 2016-07-11:
@bogdanm I get this kind of errors:
sl0(): transmit queue 0 timed out
sl0: transmit timed out, bad line quality ?

This is the trace on raspberry pi side. I have checked and verified the pin mapping. That is correct. I am using baudrate 921600

hasnainvirk, 2016-07-11:
@bogdanm I have a fix for your 'fix'. But I can't make push to k64f_flow_control branch as I don't have permissions.

bogdanm, 2016-07-12:
Closing this, since it was fixed by #2140 according to @hasnainvirk.

closed 2016-07-12 by bogdanm