ARMmbed/mbed-os #2118: RTOS - Cortex M4 without FPU - fails to compile for ARM

0xc0170, 2016-07-07:
It was fixed for GCC_ARM but we overlooked there is still problem with ARM, originally reported here: https://github.com/mbedmicro/mbed/issues/612

Error msg:
```
n opcode 'VSTMDBEQ', maybe wrong target CPU?
  167 000000ac  VSTMDBEQ R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 187: Error: A1854E: Unknown opcode 'VLDMIANE', maybe wrong target CPU?
  187 000000e0  VLDMIANE R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 250: Error: A1854E: Unknown opcode 'VSTMDBEQ', maybe wrong target CPU?
  250 00000136  VSTMDBEQ R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 269: Error: A1854E: Unknown opcode 'VLDMIANE', maybe wrong target CPU?
  269 0000016a  VLDMIANE R12!,{S16-S31}
4 Errors, 0 Warnings
C:\Code\git_repo\github\mbed\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\HAL_CM4.c: 4 warnings, 0 errors
"C:/Code/git_repo/github/mbed/.build/rtos/TARGET_CORTEX_M/rt_HAL_CM.h", line 209: Warning:  #3731-D: intrinsic is deprecated
"C:/Code/git_repo/github/mbed/.build/rtos/TARGET_CORTEX_M/rt_HAL_CM.h", line 212: Warning:  #3731-D: intrinsic is deprecated
"C:/Code/git_repo/github/mbed/.build/rtos/TARGET_CORTEX_M/rt_HAL_CM.h", line 214: Warning:  #3731-D: intrinsic is deprecated
"C:/Code/git_repo/github/mbed/.build/rtos/TARGET_CORTEX_M/rt_HAL_CM.h", line 216: Warning:  #3731-D: intrinsic is deprecated
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 167: Error: A1854E: Unknown opcode 'VSTMDBEQ', maybe wrong target CPU?
  167 000000ac  VSTMDBEQ R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 187: Error: A1854E: Unknown opcode 'VLDMIANE', maybe wrong target CPU?
  187 000000e0  VLDMIANE R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 250: Error: A1854E: Unknown opcode 'VSTMDBEQ', maybe wrong target CPU?
  250 00000136  VSTMDBEQ R12!,{S16-S31}
"mbed\\rtos\\rtx\\TARGET_CORTEX_M\\TARGET_RTOS_M4_M7\\TOOLCHAIN_ARM\\HAL_CM4.c", line 269: Error: A1854E: Unknown opcode 'VLDMIANE', maybe wrong target CPU?
  269 0000016a  VLDMIANE R12!,{S16-S31}
4 Errors, 0 Warnings
```

Fix:
```
diff --git a/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/HAL_CM4.c b/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/Hindex e1366a0..ee8ff8e 100644
--- a/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/HAL_CM4.c
+++ b/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM/HAL_CM4.c
@@ -164,7 +164,9 @@ SVC_Handler_Veneer

 SVC_ContextSave
         TST     LR,#0x10                ; is it extended frame?
+#ifdef __FPU_PRESENT
         VSTMDBEQ R12!,{S16-S31}         ; yes, stack also VFP hi-regs
+#endif
         MOVEQ   R0,#0x01                ; os_tsk->stack_frame val
         MOVNE   R0,#0x00
         STRB    R0,[R1,#TCB_STACKF]     ; os_tsk.run->stack_frame = val
@@ -184,7 +186,9 @@ SVC_ContextRestore
         CMP     R0,#0                   ; Basic/Extended Stack Frame
         MVNEQ   LR,#:NOT:0xFFFFFFFD     ; set EXC_RETURN value
         MVNNE   LR,#:NOT:0xFFFFFFED
+#ifdef __FPU_PRESENT
         VLDMIANE R12!,{S16-S31}         ; restore VFP hi-registers
+#endif
         MSR     PSP,R12                 ; Write PSP

 SVC_Exit
@@ -247,7 +251,9 @@ Sys_Switch

         MRS     R12,PSP                 ; Read PSP
         TST     LR,#0x10                ; is it extended frame?
+#ifdef __FPU_PRESENT
         VSTMDBEQ R12!,{S16-S31}         ; yes, stack also VFP hi-regs
+#endif
         MOVEQ   R0,#0x01                ; os_tsk->stack_frame val
         MOVNE   R0,#0x00
         STRB    R0,[R1,#TCB_STACKF]     ; os_tsk.run->stack_frame = val
@@ -266,7 +272,9 @@ Sys_Switch
         CMP     R0,#0                   ; Basic/Extended Stack Frame
         MVNEQ   LR,#:NOT:0xFFFFFFFD     ; set EXC_RETURN value
         MVNNE   LR,#:NOT:0xFFFFFFED
+#ifdef __FPU_PRESENT
         VLDMIANE R12!,{S16-S31}         ; restore VFP hi-regs
+#endif
         MSR     PSP,R12                 ; Write PSP

 Sys_Exit
```

duff2013, 2016-08-07:
I forked a copy [here](https://github.com/duff2013/mbed-os) and made the fix referenced in this issue and was able to compile and upload the blinky example for the Teensy 3.2. It seems to work as expected, i even added second Thread to the blinky example and it worked too.

sg-, 2017-01-16:
Should be resolved.

closed 2017-01-16 by sg-