ARMmbed/mbed-os #2124: mbed_sdk_init called twice on IAR without RTOS.

jamike, 2016-07-08:
1-mbed_sdk_init is called after __iar_data_init3 (zero initilaized and sdata recopy from flash)
2 mbed_sdk_init is called in __iar_argc_argv : this call can be removed

0xc0170, 2016-07-08:
It's first called in the cmain.S file: https://github.com/mbedmicro/mbed/blob/03d8ea8e96051c0fa2d0a081f9f84b9c86d9ffb2/hal/targets/cmsis/TOOLCHAIN_IAR/cmain.S#L53

Second time: https://github.com/mbedmicro/mbed/blob/8e8f389d3a10724626ce0eec4c14c9ff6cde79e0/hal/common/retarget.cpp#L548

I would remove it from cmain.S, as in retarget most of toolchains have it there (mbed sdk init call), not to bring confusion that we missed sdk_init call. However, looking at the PR https://github.com/mbedmicro/mbed/pull/780 and the code in cmain.s, there's an intention why mbed sdk init is called.

Please send a fix. 

cc @c1728p9

jamike, 2016-07-08:
__iar_dynamic_initialization  (  initialize C++ global object) is just called after this 1st call to mbed_sdk_init..
if we remove this 1st call to mbed_sdk_init, and the initiliazed C++ needs access to hardware IP which  relies on SysCoreClock value for a correct hardware init, it will not work. that's why I propose to remove the call from  ___iar_argc_argv.
On the other if we look at the other toolchain  init (except gcc_arm :in which the zero initialized and recopy of sdata is done at start up, before SystemInit which set SystemCoreClock correctly for ever without the need to call SystemCoreClockupdate in mbed_sdk_init ) :ARM , uARM call mbed_sdk_init tool late (after C++ initialization : __cpp_initialize__aeabi_)

0xc0170, 2016-07-21:
> On the other if we look at the other toolchain init (except gcc_arm :in which the zero initialized and recopy of sdata is done at start up, before SystemInit which set SystemCoreClock correctly for [...]

this shall be another issue, where we can continue with how to unify startup.

This was resolved via https://github.com/mbedmicro/mbed/pull/2129

closed 2016-07-21 by 0xc0170