ARMmbed/mbed-os #2126: Critical - core_util_atomic_decr/incr are undefined

0xc0170, 2016-07-08:
Undefined functions ``core_util_atomic_decr/incr``. There's no function with a declaration to match: ``core_util_atomic_decr((uintptr_t)valuePtr, (uintptr_t)delta)`` . I would assume that missing functions should be implemented, and can't be on top of current functions because they return a value, and these functions return void pointer. The API documentation should state that (be more specific), not ``The new decremented value.`` as the rest of API.

Reference: #2054

@geky

LMESTM, 2016-07-08:
I'm facing this issue as well on top of today's mbed/master ...  it seems to happen "only" on cortex-M0/M0+ targets.
Has this pass through CI compilation ?

geky, 2016-07-08:
Pr up: #2131. It was a last minute change and went unnoticed since GCC only emits a warning. 

> can't be on top of current functions because they return a value, and these functions return void pointer

That is my bad, I was missing the warning emitted for differing return values. I'll fix this in a separate pr once #2131 is merged.

geky, 2016-07-08:
Due to IAR errors I went ahead and fixed the casting issue @0xc0170 noticed on #2131. Let me know if there's still an issue.

geky, 2016-07-11:
Now that #2131 is merged, is this still an issue?

closed 2017-01-16 by sg-