ARMmbed/mbed-os #2132: NRF51 boards without 32K crystal fails to boot

c1728p9, 2016-07-08:
On the nrf51 the function ``os_tick_init`` in ``us_ticker.c`` initializes the low frequency clock to use a 32 KHz crystal and waits for it to stabilize.  If this crystal is not present the the device will hang forever waiting for stabilization.  One board which exhibits this problem is the ``micro:bit``.

To support all boards there needs to be some mechanism to select the internal RC oscillator if no external crystal is present.

As a test, selecting the internal RC oscillator allows the ``micro:bit`` to boot and run.

Before:
```
int os_tick_init (void)
{
    NRF_CLOCK->LFCLKSRC            = (CLOCK_LFCLKSRC_SRC_Xtal << CLOCK_LFCLKSRC_SRC_Pos);
...
```
With:
```
int os_tick_init (void)
{
    NRF_CLOCK->LFCLKSRC            = (CLOCK_LFCLKSRC_SRC_RC << CLOCK_LFCLKSRC_SRC_Pos);
...
```

c1728p9, 2016-07-08:
CC: @0xc0170, @pan-

0xc0170, 2016-07-09:
Good catch ! Should nrf51 targets define if there's external RC or not? I'll talk to @pan

nvlsianpu, 2016-07-19:
I will resolve this by adding clock configuration using mbed configuration system (mbed_app.json...).

nvlsianpu, 2016-07-19:
https://github.com/ARMmbed/mbed-nordic/pull/24

closed 2016-07-22 by sg-