ARMmbed/mbed-os #2135: project.py

ohagendorf, 2016-07-09:
Some commits before it was possible to call:
```
project.py -S
```
to get the supported matrix of MCUs and IDEs.
Now I get:
```
usage: project.py [-h] -m MCU -i IDE [-c] [-p PROGRAM] [-n PROGRAM] [-b] [-L]
                  [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
project.py: error: argument -m/--mcu is required
```
and have to call e.g.
```
project.py -S -m ARCH_MAX -i uvision5
```
to get this tables.
The same error is generated with :
```
project.py -L
```

build.py and make.py with option -S are working as before.

ohagendorf, 2016-07-11:
I think it was introduced in PR #2010
Maybe @theotherjimmy can help.

theotherjimmy, 2016-07-11:
Yep. On it.

PR #2141 (2016-07-14): [Project.py] Make -i and -m optional
Resolves #2135
commit cebe292c (2016-07-14): [Project.py] Make -i and -m optional

Resolves #2135
closed 2016-07-14 by sg-