ARMmbed/mbed-os #2143: MultiTech Dragonfly: Export project to IAR, the binary file cannot run on it

hxingithub, 2016-07-11:
Hi, there,

I am trying some example programs on the MultiTech Dragonfly. When I use the online compiler, and get the binary files to download them into the board by USD Disk mode, they work well. but I exported these projects to my IAR V7.4 compiler, and built these projects to binary files, downloaded them into the board as the same way, there were nothing displaying on the serial terminal. However I can use the IAR download and debug, these programs can run well.

I do not know what differences are between these binary files from online compiler and my IAR compiler.

Thanks, Henry

0xc0170, 2016-07-12:
This might be specific to mbed online IDE, not for this repository. Please try tools/project.py if you can export to IAR simple blinky "MBED_10" program.

To build use ``tools/build.py`` script, to export use ``tools/project.py``.

mfiore02, 2016-07-12:
@hxingithub the Dragonfly has a bootloader that gets automagically combined with the application when using the mbed workspace tools build tools (online compiler and offline). When you export to uvision or iar, however, this isn't the case. If you check out the linker files for the different toolchains, you'll see that the application is being built to run from a 64kB offset in flash instead of from the beginning of flash. The bootloader occupies that first 64kB of space.
You'll have to update the linker file or manually combine the bootloader with the application for things to work right. It's also possible to write up a post-build script for iar to use.

hxingithub, 2016-07-13:
@0xc0170 @mfiore02 

Thanks for reply.
But I checked the link file for the IAR, it has reserved the 64KB offset for the flash:
/* [ROM = 512kb = 0x80000] */
define symbol __intvec_start__     = 0x08010000;    /* 0x10000 is 64kB offset */
define symbol __region_ROM_start__ = 0x08010000;
define symbol __region_ROM_end__   = 0x0807FFFF;

That is confused.

mfiore02, 2016-07-13:
@hxingithub that's what I was trying to explain. If you're building an app in IAR and not using the bootloader, you'll need to change the linker script to point to the start of flash instead of a 64kB offset since nothing will be occupying that space. Or you need to add the bootloader to your app before flashing the Dragonfly.

hxingithub, 2016-07-13:
@mfiore02 Hi Mike,

Thanks. I modified the start address and it works well.

Regards,
Henry

hxingithub, 2016-07-13:
Change linker file to:
define symbol intvec_start = 0x08000000; 
define symbol region_ROM_start = 0x08000000;
define symbol region_ROM_end = 0x0807FFFF;

closed 2016-07-13 by hxingithub