ARMmbed/mbed-os #2147: Multitech Mdot NodeRed HTTP Rquest Error

eren23, 2016-07-12:
Im having trouble when im trying to send json or any string data to node.js server.

My error is: 
"ENNETUNREACH"
This appears under the httprequest node.

     There is no other explanation about the error in the page that im working on it.

     This is a really Urgent situation so if anyone have any ideas about what is wrong please let me know about your opinnions.

Thanks in Advice.

0xc0170, 2016-07-12:
cc @geky @c1728p9

eren23, 2016-07-12:
here is the error

http://prntscr.com/brwgof

mfiore02, 2016-07-12:
@eren23 you should open a support ticket with MultiTech. I don't think you'll get much support for a Node RED issue here.

http://support.multitech.com/

eren23, 2016-07-13:
@mfiore02 mate i did it already but they are kind a slow so decided to try my chance in here xd.

closed 2017-04-15 by eren23