ARMmbed/mbed-os #2170: [GCC] Multiple optimization issue.

dinau, 2016-07-14:
I tried to generate Makefile project with mbed-lib ver.122
as follows:
```
python tools/build.py -m NUCLEO_F030R8 -t GCC_ARM -c -v
python tools/project.py -m NUCLEO_F030R8 -i gcc_arm -n MBED_11
```
In the Makefile generated, multiple optimization options by default, 
"-O2" and "-Os" are both activated during compilation.

"-O2" option should be removed.

dinau

0xc0170, 2016-07-14:
> I tried to generate Makefile project with mbed-lib ver.122

using release branch?

cc @theotherjimmy

dinau, 2016-07-14:
@0xc0170 ,
>using release branch?

I just noticed that there are two commit messages about "mbed lib revision - 122" in git repository.
I will check "release branch".

0xc0170, 2016-07-14:
What I meant, on the master there's https://github.com/mbedmicro/mbed/blob/master/hal/api/mbed.h#L19 , v121, where did you get 122 ? :)

dinau, 2016-07-14:
I confirmed that above issue is existing using release branch.

dinau

0xc0170, 2016-07-14:
> I confirmed that above issue is existing using release branch.

Can you please reference the file?

dinau, 2016-07-14:
>Can you please reference the file?
Now my local repository is broken by accident.  :-(
I'll be back later.

zgoda, 2016-07-15:
On master for -t BLUEPILL_F103C8 and -i GCC_ARM I get only -O2 in CC_FLAGS:

```
CC_FLAGS = -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m3 -mthumb -O2 -std=gnu99 -include mbed_config.h -MMD -MP
```

Then there is a DEBUG flag check:

```
ifeq ($(DEBUG), 1)
  CC_FLAGS += -DDEBUG -O0
else
  CC_FLAGS += -DNDEBUG -Os
endif
```

Complete command line:

```
$ python tools/build.py -t GCC_ARM -m BLUEPILL_F103C8 -c -v
$ python tools/project.py -m BLUEPILL_F103C8 -i GCC_ARM -n MBED_11
```

dinau, 2016-07-15:
Thank you @zgoda,
I executed command line you showed and then executed Makefile generated. 
Below is part of command line "make" output includes both "-O2" and "-Os" .
```
arm-none-eabi-gcc  -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m3 -mthumb -O2 -std=gnu99 -include mbed_config.h -MMD -MP -DNDEBUG -Os -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DTARGET_BLUEPILL_F103C8 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DTARGET_STM32F103C8 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DTARGET_RELEASE -DTARGET_STM -DMBED_BUILD_TIMESTAMP=1468583760.43 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M3  -I../. -I.././env -I.././hal -I.././hal/api -I.././hal/common -I.././hal/hal -I.././hal/hal/storage_abstraction -I.././hal/targets -I.././hal/targets/cmsis -I.././hal/targets/cmsis/TARGET_STM -I.././hal/targets/cmsis/TARGET_STM/TARGET_STM32F1 -I.././hal/targets/cmsis/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8 -I.././hal/targets/cmsis/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8/TOOLCHAIN_GCC_ARM -I.././hal/targets/hal -I.././hal/targets/hal/TARGET_STM -I.././hal/targets/hal/TARGET_STM/TARGET_STM32F1 -I.././hal/targets/hal/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8  -o hal/common/assert.o ../hal/common/assert.c
```
@0xc0170
I used "master branch",
https://github.com/mbedmicro/mbed/tree/3d20b10b4d77d7820458d371dbed7641acccc671

@0xc0170
By the way,
"mbed lib revision - 122" is here.
https://github.com/mbedmicro/mbed/commits/24916505e10b69c60469629ee3ea946eac61cebf
git SHA: 24916505e10b69c60469629ee3ea946eac61cebf

dinau

zgoda, 2016-07-15:
But in fact this is not a problem with GCC because as stated in man:

> If you use multiple -O options, with or without level numbers, the last such option is the one that is effective.

So effectively we get -Os because this is the last option added.

dinau, 2016-07-15:
>If you use multiple -O options, with or without level numbers, the last such option is the one that is effective.

Yes I know that behavior, 
but many user will have confusion if two options exist.

sg-, 2017-01-16:
@dinau has this been resolved? @theotherjimmy ??

theotherjimmy, 2017-01-17:
Yes

closed 2017-01-17 by sg-