ARMmbed/mbed-os #2183: project.py fail

svastm, 2016-07-18:
After #2176,  i'm not able to use `project.py`.

I have the following error:
```
> ..\mbed\tools\project.py -c -m NUCLEO_L432KC -p 0 -i uvision5
Generating the project, please wait (it takes few minutes)...
Traceback (most recent call last):
  File "..\mbed\tools\project.py", line 144, in <module>
    tmp_path, report = export(project_dir, project_name, ide, mcu, project_dir[0], project_temp, clean=clean, make_zip=zip, extra_symbols=lib_symbols, sources_relative=sources_relative)
  File "C:\Users\xxx\github\mbed\tools\export\__init__.py", line 97, in export
    if target not in Exporter.TARGETS or Exporter.TOOLCHAIN not in TARGET_MAP[target].supported_toolchains:
TypeError: argument of type 'property' is not iterable

Finished.
```

if `TARGETS` is not anymore a class attribute, the `Exporter` variable should refer to an instance and not a class.

0xc0170, 2016-07-18:
Thanks for reporting, I'll have a look shortly

0xc0170, 2016-07-18:
Working on the fix, I was able to reproduce it

commit f80b62ed (2016-07-19): exporter - fix TARGETS property via descriptors

Fixes #2183 issue.
closed 2016-07-19 by bogdanm