ARMmbed/mbed-os #2201: Inconsistent way to set RTOS single threaded mode

pan-, 2016-07-21:
If the target parameter `default_build` is set to `small` then when GCC is used the result will be a single threaded application.
If the same program is compiled with the `ARM` toolchain then single thread mode will not be activated. 
Instead the user have to use `uARM`. 

This is inconsistent, either we should have a uGCC target or the ARM toolchain enable single thread mode when `default_build` is equal to `small`.

0xc0170, 2016-07-21:
We should deprecate uARM, start using ``default_build``, to be consistent with other toolchains (gcc arm in this case)

bridadan, 2017-01-17:
I was looking into this a bit, and it looks like GCC is the only toolchain that's actually modifying the standard library based on `default_lib` (used to be `default_build`)? It doesn't really look like uARM is doing this more any more, or at least I can't find where in the code it does this. And IAR doesn't seem to be doing anything with it either, though I don't think we ever supported this.

@theotherjimmy does this sound right to you?

theotherjimmy, 2017-01-18:
Yes. I remember uARM being always small, ARM being always big, IAR being always big, and GCC heeding the `default_*` data.

bridadan, 2017-01-18:
@theotherjimmy 

>  I remember uARM being always small

I agree, but I can't find for the life of me where this actually happens in the code. Where it actually switches the lib I mean. Can you?

theotherjimmy, 2017-01-18:
Huh. I can't either...

theotherjimmy, 2017-01-18:
So, moving forward with this issue, lets make ARM heed the `default_*` data.

bridadan, 2017-01-18:
@theotherjimmy yup, and make `uARM` an alias to having `ARM` with `default_lib: small` (and perhaps adding a deprecation notice for `uARM`, though not sure about that one actually since it has implications for mbed 2)

theotherjimmy, 2017-01-18:
@bridadan you want uARM to change the target object passed into it to have `default_lib = small`. That's a bit strange.

bridadan, 2017-01-18:
@theotherjimmy uARM is only relevant for mbed OS 2. This is legacy behavior that we need to keep supporting I believe.

theotherjimmy, 2017-01-19:
Sounds like uARM is here to stay then.

bridadan, 2017-01-19:
...for the time being anyway :smiling_imp:. I'm honestly just not a big fan of it, seems like it really should be a build profile thing instead of changing the whole toolchain.

theotherjimmy, 2017-01-19:
We'll see. Maybe we can sort it out soon.

closed 2017-10-27 by ChiefBureaucraticOfficer