ARMmbed/mbed-os #2206: mbed test --run fails for all tests

simonqhughes, 2016-07-21:
##### Summary

The following command appears not to work in mbedmicro/mbed

```
(mx4_venv1) simhug01@E107851:/d/datastore/public/jobs/yr2016/2253/sdh_dev_mx4/mbed_sqh1$ mbed test --run  2>&1 | tee mbed_sqh1_mbedgt_run_without_stoage_Os_20160721_1033.txt
```

##### Details

In order to setup the development environment I did the following:

- Setup new virtualenv, installed mbed-cli v0.8.7
- Cloned mbedmicro/mbed. 
- Pip install the mbedmicro/mbed requirements.txt
- Pip install pyelftools
- Mbed test –compile in mbedmiro/mbed. The tests built fine.
- Mbed test –run on my board. See attached log file, which has TIMEOUT/ERROR, like you observe for CFSTORE tests

The tests were built with this command:
```
(mx4_venv1) simhug01@E107851:/d/datastore/public/jobs/yr2016/2253/sdh_dev_mx4/mbed_sqh1$ mbed test -v --compile -m K64F -t GCC_ARM 2>&1 | tee mbed_sqh1_build_tests_with_storage_Os_20160721_1033.txt
```

However, all the tests fail e.g.

```
mbedgt: test on hardware with target id: 0240000029304e450023500878a3001df131000097969900
mbedgt: test suite 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... TIMEOUT in 20.09 sec
	test case: 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... ERROR in 20.09 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found: two directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
	no test case report present, assuming test suite to be a single test case!
	test suite: tests-mbedmicro-rtos-mbed-queue
	test case: tests-mbedmicro-rtos-mbed-queue
mbedgt: test on hardware with target id: 0240000029304e450023500878a3001df131000097969900
mbedgt: test suite 'tests-mbedmicro-rtos-mbed-queue' ................................................. TIMEOUT in 20.38 sec
	test case: 'tests-mbedmicro-rtos-mbed-queue' ................................................. ERROR in 20.38 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found: two directory levels above image path checked

```

@bridadan 

[mbed_sqh1_mbedgt_run_without_storage_Os_20160721_1033.txt](https://github.com/mbedmicro/mbed/files/375956/mbed_sqh1_mbedgt_run_without_storage_Os_20160721_1033.txt)

bridadan, 2016-07-21:
Hi @simonqhughes, could you run `mbed test` with `v` added? The RTOS tests were working last time I tried it, I'm wondering if there's an issue with the connection to our board

bridadan, 2016-07-21:
I followed your instructions and the tests you mentioned as timing out are working fine for me. Here's more info about my environment:

git commit: db60d6a7f72613a53df4a19b3d73849819013680
OS: Windows 7

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

And the log from my run:

```
$ mbed test --run -m K64F -t GCC_ARM
mbedgt: greentea test automation tool ver. 1.1.1
mbedgt: test specification file 'C:\Users\bridan01\Documents\dev\m_mbed\mbed\.build/tests\K64F\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Users\bridan01\Documents\dev\m_mbed\mbed\.build/tests\K64F\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | K64F          | K64F[0]              | COM255      | D:          | 0240000028884e450037700f6bf0000d8021000097969900 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | K64F          | K64F[0]              | COM255:9600 | D:          | 0240000028884e450037700f6bf0000d8021000097969900 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: running 40 tests for platform 'K64F' and toolchain 'GCC_ARM'
        use 1 instance for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000028884e450037700f6bf0000d8021000097969900
mbedgt: test suite 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... OK in 11.98 sec
        test case: 'dummy test' ...................................................................... OK in 0.04 sec
        test case: 'dummy test 2' .................................................................... OK in 0.04 sec
mbedgt: test case summary: 2 passes, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-mbedmicro-rtos-mbed-queue
        test case: tests-mbedmicro-rtos-mbed-queue
mbedgt: test on hardware with target id: 0240000028884e450037700f6bf0000d8021000097969900
mbedgt: test suite 'tests-mbedmicro-rtos-mbed-queue' ................................................. OK in 13.14 sec
        test case: 'tests-mbedmicro-rtos-mbed-queue' ................................................. OK in 13.14 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: mbed-host-test-runner: started
```

bridadan, 2016-07-29:
I'm going to close this for now since I can't reproduce this. @simonqhughes if you still have trouble setting up a testing environment please reopen this so we can troubleshoot.

closed 2016-07-29 by bridadan

zackwwu, 2017-07-28:
@simonqhughes  i encountered exactly the same issue, have you fixed this problem? And how did you mange to resolve it. Thanks!

bridadan, 2017-07-28:
@zackwwu Can you post the verbose log from your test run?