ARMmbed/mbed-os #2209: target_overrides only allows to add extra features, not labels/macros/device_has

AlessandroA, 2016-07-21:
The `target_overrides` config option only allows to add/remove custom features (`FEATURE_*`). Other symbols can only be overridden (the symbol needs to be there in the first place).

This means that it is not possible for `target_overrides` to add/remove a new label (`TARGET_*`), a new custom macro or a new `DEVICE_HAS_*` macro.

**How to reproduce**

If in an application you override the K64F target, this is allowed:

```json
{
    "target_overrides": {
        "K64F": {
            "target.features_add": ["UVISOR"]
        }
    }
}
```

but this is not (compile error):

```json
{
    "target_overrides": {
        "K64F": {
            "target.features_add": ["UVISOR"],
            "target.extra_labels_add": ["UVISOR_SUPPORTED"]
        }
    }
}
```

**Additional notes**

This feature was originally available for custom targets (which BTW are not working any more, see #2196 and [this related issue](https://github.com/ARMmbed/mbed-cli/issues/281)).

In the case of the uVisor hello-world example and other similar applications, though, it is not ideal to define custom targets. `target_overrides` are more natural.

This is a blocker for the release of the uVisor hello-world application.

@screamerbg @meriac @mjs-arm

sg-, 2016-07-21:
@geky please review

geky, 2016-07-21:
I have a branch that should properly handle cumulative overrides now:
https://github.com/mbedmicro/mbed/compare/master...geky:tools-cumulative-overrides

@screamerbg, @theotherjimmy, does this seem reasonable?

I don't know if it is correctly handling the uVisor supported label though, there are several special cases in the build tools ([example](https://github.com/mbedmicro/mbed/blob/51540d08d57e27441b8d088a7191435940a0123c/tools/toolchains/__init__.py#L295)).

geky, 2016-07-21:
Moved to pr: https://github.com/mbedmicro/mbed/pull/2215

sg-, 2016-09-22:
@AlessandroA is this resolved?

AlessandroA, 2016-09-22:
@sg- Yes, it was fixed by the linked PR. Closing the issue.

closed 2016-09-22 by AlessandroA