ARMmbed/mbed-os #2212: The config system should support configuration "profiles"

AlessandroA, 2016-07-21:
The configuration system should provide a way to have configuration *profiles*.

@screamerbg @sg- @meriac @mjs-arm

===

A profile is just like any configuration file (JSON), but can co-exist with other configuration files in the same app, library, target or test – without being picked up by default.

If the user does not specify a profile, then no profile is used.

The build system would provide a new option to specify the profile. Example:

```bash
$ mbed compile -m K64F -t GCC_ARM -p uvisor_supported_but_disabled.json
```

**Use cases**

Profiles could be used by libraries to provide different flavors. For example, a library could be compiled optimizing for speed or for space (different build options), or could provide some readymade configurations that users can choose from.

**Requirement for uVisor**

Configuration profiles (or an equivalent alternative) are a requirement for uVisor testing.

Each uVisor test case must be run against a configuration hypermatrix. One axis of this matrix is a set of symbols that defines a uVisor flavor.

We would auto-generate all the (feasible) uVisor flavors and package them as profiles. Each test could then be run against multiple profiles:

```bash
$ mbed test -m K64F -t GCC_ARM -n tests-case1 -p uvisor_supported_but_disabled.json
$ mbed test -m K64F -t GCC_ARM -n tests-case1 -p uvisor_supported_and_enabled.json
$ mbed test -m K64F -t GCC_ARM -n tests-case1 -p uvisor_unsupported.json
$ ...
```

[This](https://gist.github.com/AlessandroA/338daf9c68a36a5b9e2bffdc1ef90cc8) is an example of how we translate the test hypermatrix into a CI-friendly script. For more information on the uVisor testing requirements see [here](https://gist.github.com/AlessandroA/2cfdf04048e3e96672b703415adc8966).

geky, 2016-07-21:
Is new vocabulary necessary? Correct me if I'm wrong, but it sounds like this requirement would be satisfied by just adding a switch to choose the config file:

``` bash
$ mbed test -m K64F -t GCC_ARM -n tests-case1 --config=uvisor_supported_but_disabled.json
```

@theotherjimmy thoughts?

theotherjimmy, 2016-07-21:
This sounds more general, and would not be difficult to implement without changing the tools API's.

AlessandroA, 2016-07-21:
@geky  I was thinking about a new name because currently a config file is exclusive and is always applied to the build.

Instead, a profile would need to be explicitly selected by the user. This also allows you to have multiple JSON files in the same folder, each one defining a new profile.

You can also implement this in the JSON file instead of in the command line tool. Something like `"config": { ... }` vs `"profile": { ... }`. I still think the semantics is different from the config as it is now, though. What do you think?

AlessandroA, 2016-07-26:
@theotherjimmy @sg- @geky Bump

If the general concept of "profile" is not worth implementing now, a way of specifying a config file from the command line would be enough to satisfy the requirement for uVisor testing (basically what @geky proposed [above](https://github.com/mbedmicro/mbed/issues/2212#issuecomment-234377471)).

IIUC a config file is not picked up automatically if it's not in the same folder as the app. I can keep all my profiles in a separate folder and select them when needed with a command line option like: `--config "profiles/profile1.json"`.

bogdanm, 2016-07-26:
> Correct me if I'm wrong, but it sounds like this requirement would be satisfied by just adding a switch to choose the config file:

I believe you are right, although I'd make that more general: add more than one config option.

```
$ mbed test -m K64F -t GCC_ARM -n tests-case1 --config=uvisor_supported_but_disabled.json --config=whatever_else.json ...
```

The `--config` arguments would be processed in the order they appear in the command line, such as that `--config` argument `n` would be able to override `--config` arguments `1, 2, ..., n-1`. We actually had people asking for this feature at some point in the past.

sg-, 2016-09-22:
@AlessandroA is this now implemented?

AlessandroA, 2016-09-22:
@sg- Yes, this is now implemented by the `--app-config` option. Thanks!

closed 2016-09-22 by AlessandroA