ARMmbed/mbed-os #2217: Bug in wait_api: us_ticker might not be initialized

pan-, 2016-07-22:
As stated in the documentation of [wait_api.h](https://github.com/mbedmicro/mbed/blob/master/hal/api/wait_api.h). 
The API can be use like this: 

```c++
DigitalOut heartbeat(LED1);

int main() {
    while (1) {
        heartbeat = 1;
        wait(0.5);
        heartbeat = 0;
        wait(0.5);
    }
}
```

In the implementation, us_ticker_read is used to wait but might not be initialized: https://github.com/mbedmicro/mbed/blob/master/hal/common/wait_api.c#L29

On implementation where `us_ticker_init` does actually initialize something needed by `us_ticker_read`, this makes  the wait_api unusable.

0xc0170, 2016-07-22:
My 5 cents, us_ticker_read() is public function and can be used anywhere (the only function to provide  time), therefore many (probably all targets) initialize ticker in the read() as well.

```
// taken 
uint32_t us_ticker_read() {
    if (!us_ticker_inited) {
        us_ticker_init();
    }

    return ~(PIT_GetCurrentTimerCount(PIT, kPIT_Chnl_1));
}
```

Having said  that, wait_api should initialze ticker or us_ticker_read() should document the intentional behavior that many of us assume and same for wait_api().

pan-, 2016-07-22:
That's what I though but it was not clear.
This means that as long as any wait API has been called the microsecond ticker will run and therefore consume power, I think the documentation should be vocal about this.

c1728p9, 2016-07-22:
@pan- you might be able to get around this issue by turning on the timer at the start of the wait and turning it off at the end (or keep a usage count in the case of multiple threads).

0xc0170, 2016-07-22:
> @pan- you might be able to get around this issue by turning on the timer at the start of the wait and turning it off at the end (or keep a usage count in the case of multiple threads).

that would be ideal (turn it on when it's needed). However, us_ticker is considered on mbed to provide 32bit timestamp, thus always running.

c1728p9, 2016-07-22:
Maybe the first call to us_ticker for an absolute timestamp could keep it on. Longer term it might be good to deprecate the API because of the negative implications and force users to create a Timer object explicitly. @pan-, @0xc0170 what are your thoughts?

sg-, 2016-09-22:
```
    if (!us_ticker_inited) {
        us_ticker_init();
    }
```
should be moved into common code

closed 2017-10-27 by ChiefBureaucraticOfficer