ARMmbed/mbed-os #2236: Path to BLE files for target NRF51822 is too long

bridadan, 2016-07-24:
When building for any NRF51822 based target on Windows, you run into the Windows path limit of 260 characters very quickly. This causes the compile step to fail during copying.

@pan- Could the necessary files be flattened somewhat? I have to build in a directory that's fairly close to the disk root to avoid this problem currently.

bridadan, 2016-07-24:
Also FYI @sg-

0xc0170, 2016-07-26:
>When building for any NRF51822 based target on Windows, you run into the Windows path limit of 260 characters very quickly

what toolchain ? didn't we fix that using intermediate files? Or is this for linking phase? please specify the command to reproduce.

bridadan, 2016-07-26:
@0xc0170 This is for any toolchain, and this occurred during the copy file step. This was the exact command that was ran in CI at the working directory `C:/Jenkins/workspace/build_matrix_wrapper/node_name=build_node_2/135`:

```
mbed test --compile -m NRF51_DK -t ARM --build-report-junit=C:/Jenkins/workspace/build_matrix_wrapper/node_name=build_node_2/135/.axes/target/NRF51_DK/toolchain/ARM/build_report_NRF51_DK_ARM.xml
```

And here's the full error:
```
22:02:31 + mbed test --compile -m NRF51_DK -t ARM --build-report-junit=C:/Jenkins/workspace/build_matrix_wrapper/node_name=build_node_2/135/.axes/target/NRF51_DK/toolchain/ARM/build_report_NRF51_DK_ARM.xml
22:02:32 Building library mbed-build (NRF51_DK, ARM)
22:02:32 Scan: 135
22:02:32 Scan: FEATURE_BLE
22:02:32 Scan: FEATURE_UVISOR
22:02:32 Scan: FEATURE_IPV4
22:02:32 Copy: test_env.h
22:02:32 Copy: unity.h
22:02:32 Copy: unity_config.h
22:02:32 Copy: unity_internals.h
22:02:32 Copy: unity_handler.h
22:02:32 Copy: utest.h
22:02:32 Copy: utest_case.h
22:02:32 Copy: utest_default_handlers.h
22:02:32 Copy: utest_harness.h
22:02:32 Copy: utest_scheduler.h
22:02:32 Copy: utest_serial.h
22:02:32 Copy: utest_shim.h
22:02:32 Copy: utest_specification.h
22:02:32 Copy: utest_stack_trace.h
22:02:32 Copy: utest_types.h
22:02:32 Copy: aes.h
22:02:32 Copy: aesni.h
22:02:32 Copy: arc4.h
22:02:32 Copy: asn1.h
22:02:32 Copy: asn1write.h
22:02:32 Copy: base64.h
22:02:32 Copy: bignum.h
22:02:32 Copy: blowfish.h
22:02:32 Copy: bn_mul.h
22:02:32 Copy: camellia.h
22:02:32 Copy: ccm.h
22:02:32 Copy: certs.h
22:02:32 Copy: check_config.h
22:02:32 Copy: cipher.h
22:02:32 Copy: cipher_internal.h
22:02:32 Copy: compat-1.3.h
22:02:32 Copy: config.h
22:02:32 Copy: ctr_drbg.h
22:02:32 Copy: debug.h
22:02:32 Copy: des.h
22:02:32 Copy: dhm.h
22:02:32 Copy: ecdh.h
22:02:32 Copy: ecdsa.h
22:02:32 Copy: ecjpake.h
22:02:32 Copy: ecp.h
22:02:32 Copy: entropy.h
22:02:32 Copy: entropy_poll.h
22:02:32 Copy: error.h
22:02:32 Copy: gcm.h
22:02:32 Copy: havege.h
22:02:32 Copy: hmac_drbg.h
22:02:32 Copy: md.h
22:02:32 Copy: md2.h
22:02:32 Copy: md4.h
22:02:32 Copy: md5.h
22:02:32 Copy: md_internal.h
22:02:32 Copy: memory_buffer_alloc.h
22:02:32 Copy: net.h
22:02:32 Copy: oid.h
22:02:32 Copy: padlock.h
22:02:32 Copy: pem.h
[... 283 lines cut ...]
22:02:32 Copy: nordic_common.h
22:02:32 Copy: nrf_assert.h
22:02:32 Copy: sdk_common.h
22:02:32 Copy: sdk_errors.h
22:02:32 Copy: sdk_mapped_flags.h
22:02:32 Copy: sdk_os.h
22:02:32 Copy: ant_stack_handler_types.h
22:02:32 Copy: ble_stack_handler_types.h
22:02:32 Copy: softdevice_handler.h
22:02:32 Copy: softdevice_handler_appsh.h
22:02:32 [Errno 2] No such file or directory: 'C:\\Jenkins\\workspace\\build_matrix_wrapper\\node_name=build_node_2\\135\\.build/tests\\NRF51_DK\\ARM\\features\\FEATURE_BLE\\targets\\TARGET_NORDIC\\TARGET_MCU_NRF51822\\nrf51-sdk\\source\\nordic_sdk\\components\\softdevice\\common\\softdevice_handler\\softdevice_handler_appsh.h'
22:02:32 Failed to build library
22:02:32 Traceback (most recent call last):
22:02:32   File "C:\Jenkins\workspace\build_matrix_wrapper\node_name=build_node_2\135\tools\test.py", line 221, in <module>
22:02:32     report_exporter.report_to_file(build_report, options.build_report_junit, test_suite_properties=build_properties)
22:02:32   File "C:\Jenkins\workspace\build_matrix_wrapper\node_name=build_node_2\135\tools\test_exporters.py", line 100, in report_to_file
22:02:32     report = self.report(test_summary_ext, test_suite_properties=test_suite_properties)
22:02:32   File "C:\Jenkins\workspace\build_matrix_wrapper\node_name=build_node_2\135\tools\test_exporters.py", line 86, in report
22:02:32     return self.exporter_junit(test_summary_ext, test_suite_properties)
22:02:32   File "C:\Jenkins\workspace\build_matrix_wrapper\node_name=build_node_2\135\tools\test_exporters.py", line 291, in exporter_junit
22:02:32     message = test_result['result']
22:02:32 KeyError: 'result'
22:02:32 [ERROR] 'result'
22:02:32 [mbed] ERROR: "python" returned error code 1.
```

toyowata, 2016-07-27:
@bridadan Just FYI, I used ```subst``` command to workaround this issue during OOB test.

```
C:\Users\toywat01\Documents\GitHub\mbed-os-example-ble>subst x: .
```

bridadan, 2016-07-27:
@TomoYamanaka I never knew about that command, good to know! Thanks a lot :)

TomoYamanaka, 2016-07-28:
@bridadan 

This is a message to Watarai-san, isn't it ? :)

bogdanm, 2016-07-28:
@bridadan: did `subst` actually help? That doesn't make sense. The `copy files` step doesn't have anything do to with the command line, since it calls `copyfile` in the `shutil` Python module, and `copyfile` shouldn't suffer from this limitation. Are you sure this isn't different problem?

pan-, 2016-07-29:
@bridadan I would like to avoid modifications in Nordic SDK layout, not because it is difficult or I'm lazy but for maintenance purposes. 

This is one of this case when we ask to *fix* the code instead of fixing the tools. 

Our *build system* is too limited to deal correctly with third parties library. 

At the moment if you want to do the perfect import of a third party library in mbed_os, you should do the following steps:
* prefix all your file name including compilation unit and private headers.
* flatten your directories (I can't imagine boost imported in mbed).
* update your code so it take account of the modifications you done.

If a library `A` has been imported into mbed and a user need to import a library `B` then you could add another step in the process: 
* update content in library `B` so it take into account changes in library `A`.

This is madness from a maintenance perspective.

bridadan, 2016-07-29:
Ah @TomoYamanaka sorry about that! There are too many people that show up when I type `@to` and I didn't have enough coffee that morning :) That was indeed for Watarai-san!

@bogdanm I haven't actually tried the `subst` command yet, but I agree with @pan- that it isn't a valid solution. I was also surprised that this cropped up during the "copy files" step because its all done within Python, not the command line. I will have have to look into this more.

@pan- I see your point about maintenance issues if you have to do lots of file manipulations. Though the prefixing of files I believe is necessary not just for our build system, but for other IDEs as well (though I admit I'm not sure which IDEs currently require this).

The path length issue has come up multiple times, I think it may be time soon to really look at this issue and try to solve it once and for all. We could use Unicode paths on Windows throughout the tools to prevent these kinds of errors.

pan-, 2016-07-31:
> @pan- I see your point about maintenance issues if you have to do lots of file manipulations. Though the prefixing of files I believe is necessary not just for our build system, but for other IDEs as well (though I admit I'm not sure which IDEs currently require this).

From my point of view, the main issue lie in the fact that the build system tries to build everything in an executable and skip the library build step. 

A library is  composed of sources files and header files (I don't include resources files in the discussion) and **dependencies**. 
During the build of the library, random path should not be included in the inclusion path otherwise you can get unexpected build results. 
After the build two kind of artifacts are available:
* A library: Which is just the collection of object files created by the compiler
* An entry point which have to be used to include library definitions, this is usually just a folder containing public headers. 

When an application is build, all those steps are skipped which lead to the issue we know:
* use object file instead of library files => Cause a failure on IAR if two objects files have the same name.
* Produce an entry point to the library which can be consumed by its dependencies: Instead the *buid system* include every single folder in the project, by doing that, every single headers is visible to every compilation unit and library. It lead to the inclusion of *random path* at compilation. Headers like `config.h` or `debug.h` are fairly common in libraries, it is fine to have those as long as all the paths possible are not included by the compiler. If the entry point of a library is the folder `libXYZ` then the content of `libXYZ` should not be directly visible to others. Instead other code should include XYZ headers with its prefix, in this case `libxyz/`.

This model has worked for decades on unix and other system, our case is not different. Why should we forced users to adapt their code while we can fix the tools ?

Just putting metadata around library layout in `mbed_lib.json` can *fix* the issues we have. There is room to put the dependencies needed, the folder exported, the include path required, etc...

closed 2016-08-03 by sg-