ARMmbed/mbed-os #2237: Problem exporting project to offline GCC Arm for Nucleo boards

gabrielrivas, 2016-07-25:
**Hardware Used:** 
Nucleo F303K8, Nucleo F446RE

**Description of the Issue**:
Today I started having problems when tried to build an exported project to the GCC Arm Embedded tool.

After tried to build the project I got these errors:
.././mbed/TARGET_NUCLEO_F303K8/stm32f303x8.h:761:29: error: expected identifier before '(' token

define CAN ((CAN_TypeDef *) CAN_BASE) ^ .././mbed/CAN.h:67:7: note: in expansion of macro 'CAN' class CAN { ^ .././mbed/TARGET_NUCLEO_F303K8/stm32f303x8.h:761:43: error: expected ')' before '*' token
define CAN ((CAN_TypeDef *) CAN_BASE) ^ .././mbed/CAN.h:67:7: note: in expansion of macro 'CAN' class CAN { ^ .././mbed/TARGET_NUCLEO_F303K8/stm32f303x8.h:761:43: error: expected ')' before '*' token
define CAN ((CAN_TypeDef *) CAN_BASE) ^ .././mbed/CAN.h:67:7: note: in expansion of macro 'CAN' class CAN { ^ ../main.cpp:12:1: error: expected '}' at end of input } ^ make[1]: * [main.o] Error 1 make: * [all] Error 2

**Workaround**
I started digging a little bit more on this issue, and compared the Makefile I got Today
with a (Previously working) Makefile I got like one month ago, and there seem to be fundamental
changes on how the Makefile/export process are being made. I changed the new Makefile to match the format and composition of the older Makefile, and tried building the project again, it built and ran as expected.

**To reproduce the Issue** 
Export any Nucleo demo ( for example: blinky) into GCC Arm embedded, and try building it offline.

0xc0170, 2016-07-25:
cc @bcostm @adustm @theotherjimmy @sarahmarshy

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-

RobsydRosca, 2016-11-07:
**Hardware Used:**
Nucleo L476RG

**Description of the Issue:**
I got the same issue, reported by gabrielrivas, importing the project Lorawan PingPong in Eclipse AC6 environment.
[image]
Here the problem seems to be the incorrect expansion of the CAN macro.
Any help or any detailed workaround is really appreciated.

Thx.

0xc0170, 2016-11-07:
That macro seems to be just info, the error is somewhere else?

cc @theotherjimmy @sarahmarshy

RobsydRosca, 2016-11-25:
At moment I don't need the CAN hal, so I defined DEVICE_CAN = 0 and excluded the code which generates the error.