ARMmbed/mbed-os #2240: Multiple Socket.h files in the build tree.

SeppoTakalo, 2016-07-25:
There is two different Socket.h headers in the build directory:
```
mbed-os$ find . -name Socket.h
./core/features/net/network-socket/Socket.h
./core/libraries/net/lwip/Socket/Socket.h
```

LwIP `Socket.h` need to be renamed, just to be safe. `Socket.h` should always point to mbed OS header.

This will break https://github.com/mbedmicro/mbed/pull/2205

geky, 2016-07-25:
I believe this is a duplicate of https://github.com/mbedmicro/mbed/issues/1939.

In an effort to move forward on a solution I put together a proposal in the discussion https://github.com/mbedmicro/mbed/issues/1939#issuecomment-235052835.

In this case, the proper way to include Socket.h without conflicts would be:
- `#include "network-socket/Socket.h"` for the nsapi socket
- `#include "lwip/Socket/Socket.h"` for the lwip socket.

Alternatively this works as along as nsapi.h is unique:
- `#include "nsapi.h"` for nsapi definitions

Let me know if you find issues with this or feel free to leave them on https://github.com/mbedmicro/mbed/issues/1939.

sg-, 2016-07-26:
please also include in (not nsapi.h but headers directly) mbed.h also needs to be guarded by a release type

geky, 2016-07-26:
@sg- is right, in the long run mbed.h will grant access to the user-facing api of the nsapi, alleviating this problem.

closed 2016-09-22 by sg-