ARMmbed/mbed-os #2242: pending ISR and may execute aufter diabling interrupt

porkybrain, 2016-07-25:
If I am interpreting [this](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0321a/BIHHFHJD.htm) correctly code resulting in the following (psydo) assembler would allow the load to happen before the pending ISR and thus causing many bugs in mbed.

``` 
str r0, ICER   #store mask into ICER (also known as NVIC_CLRENA on other chips)
ldr r1, r2       #load from SFR
<---- interrupt happens here because clear has not propagated yet
#use r1 although it is outdated due to modifications in the ISR
```

[this](https://github.com/mbedmicro/mbed/blob/a3a4c78ea01194fa26c603c8f3788c47a168b4fc/hal/targets/hal/TARGET_Silicon_Labs/TARGET_EFM32/gpio_irq_api.c#L122) would be one such example of a possible bug (after optimization of course)

0xc0170, 2016-07-29:
> If I am interpreting this correctly code resulting in the following (psydo) assembler would allow the load to happen before the pending ISR and thus causing many bugs in mbed.

the above link is broken?

cc @c1728p9 @bogdanm @geky

porkybrain, 2016-07-31:
thanks martin, link should work now.

porkybrain, 2016-08-03:
I wonder what `ARM Internal Ref: IOTMORF-118` means, if you guys are serious about removing race conditions I would be glad to offer some pointers, I have stopped reporting most of them because no one seemed to be interested up 'till now. The USB stack for example is full of them, many which actually occur in typical user code on a regular basis.

sg-, 2016-09-22:
@porkybrain Thanks

We'll start looking at these and if you want to help out, more than happy to collaborate!

c1728p9, 2016-11-14:
Hi @porkybrain, thanks for reporting.  I opened an issue to track this in CMSIS here: https://github.com/ARM-software/CMSIS_5/issues/110

porkybrain, 2016-11-21:
nice to see a definitive answer, after trying to actually reproduce the problem on hardware was not successful I was thinking it may just be an issue of me reading the spec in an overly paranoid way ;)

PR #3282 (2016-11-21): Ensure isr cannot occur after NVIC_DisableIRQ
Add data and instruction synchronization barriers to prevent interrupts from occurring after NVIC_DisableIRQ is called. This is a backport of changes made in CMSIS_5 [here](https://github.com/ARM-software/CMSIS_5/issues/110). This fixes #2242.
closed 2016-11-21 by 0xc0170