ARMmbed/mbed-os #2251: can_api.c

kiransajeev85, 2016-07-26:
CANMsg assumes data length as 8 bytes.
There could be response messages which exceed 8 byte data. This needs support.

// Type definition to hold a CAN message
struct CANMsg {
    **unsigned char data[8];        // CAN Message Data Bytes 0-7**
};

kiransajeev85, 2016-07-29:
Any updates on when this issue will be handled ?

kiransajeev85, 2016-08-01:
Could you explain meaning of internal reference. Are we not able to track it here?

sg-, 2016-09-22:
If you have a fix please send a patch.

kiransajeev85, 2016-09-23:
Hi sg,

We do not have a fix at the moment but we can take it up if no solution is available already. Automotive OBDII standard defines message fields having more than 8 bytes and no limit is imposed as such. For example Vehicle Identification Number field is 17 bytes.

Regards,
Kiran

kiransajeev85, 2016-09-26:
Following changes are identified

```
      \mbed-os-master\hal\targets\hal\TARGET_NXP\TARGET_LPC176X\can_api.c
	Line 62:     unsigned char data[20];        // CAN Message Data Bytes 0-19 :: KIRAN UPDATED
	Line 350:     m.dlc  = msg.len; //& 0xF; // KIRAN UPDATED
```

closed 2017-10-27 by ChiefBureaucraticOfficer