ARMmbed/mbed-os #2252: ARM_BEETLE_SOC target does not support IAR

MarceloSalazar, 2016-07-26:
Support for IAR toolchain is not available in RC#1
https://github.com/mbedmicro/mbed/tree/mbed-os-5.1.0-rc1

@fvincenzo @screamerbg

fvincenzo, 2016-08-12:
IAR support has been added on Beetle platform PR https://github.com/ARMmbed/mbed-os/pull/2427 
```
mbedgt: test suite report:
+--------------------+----------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| target             | platform_name  | test suite                                                                           | result | elapsed_time (sec) | copy_method |
+--------------------+----------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK     | 18.72              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK     | 18.67              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK     | 20.54              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK     | 26.7               | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK     | 20.72              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK     | 18.7               | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK     | 19.19              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK     | 19.33              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK     | 19.77              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK     | 19.54              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK     | 20.15              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK     | 18.41              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK     | 18.56              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK     | 18.57              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-rtos-rtx-target_cortex_m-tests-memory-heap_and_stack                         | OK     | 18.04              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-integration-basic                                                      | OK     | 17.82              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-integration-threaded_blinky                                            | OK     | 37.76              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-c_strings                                                 | OK     | 20.55              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-callback                                                  | OK     | 25.59              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-dev_null                                                  | OK     | 20.53              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-echo                                                      | OK     | 29.59              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-generic_tests                                             | OK     | 19.14              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-rtc                                                       | OK     | 29.01              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-stl_features                                              | OK     | 19.52              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker                                                    | OK     | 49.76              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-timeout                                                   | OK     | 29.9               | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-wait_us                                                   | OK     | 28.31              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK     | 20.45              | shell       |
[... cut ...]
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | OK     | 24.11              | shell       |
| ARM_BEETLE_SOC-IAR | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK     | 32.75              | shell       |
+--------------------+----------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 41 OK
```
I think this issue can be closed.

sg-, 2016-09-22:
Fixed

closed 2016-09-22 by sg-