ARMmbed/mbed-os #2254: [exporter] Keil/MDK project does not have correct flash algorithm for NUCLEO_F411RE

MarceloSalazar, 2016-07-26:
Tested blinky and used mbed CLI to generate a Keil project.
Attempted to download code to NUCLEO_F411RE target and found out doesn't work because it uses a 256K flash algorithm instead of the 512K.

@0xc0170

0xc0170, 2016-07-26:
@MarceloSalazar version of progendef ?

``progendef --version`` 

This is the latest definition for that mcu: https://github.com/project-generator/project_generator_definitions/blob/master/project_generator_definitions/mcu/st/stm32f411re.yaml

cc @bcostm @adustm @ohagendorf

MarceloSalazar, 2016-07-26:
```progendef --version
0.2.27
```
It would be great if we could have this fixed soon (before we start OOB for RC#2)

ohagendorf, 2016-07-26:
I can have a look at it not before the next weekend, I'm very sorry.
I'm currently only reading here.

0xc0170, 2016-07-27:
I can confirm the device flash algo is 256K. however, the target seems to have proper definition. I believe this is connected to uvopt file.

This was fixed in progen v0.9.8 thanks to @ohagendorf. The corrections were done however only for uvision5, therefore ``mbed export -i uvision5 -m NUCLEO_F411RE`` should work.

I'll try to fix this for uvision as well

0xc0170, 2016-07-27:
I got it working for f411.

@MarceloSalazar can you test these 2 PR (both of them to be applied): 
https://github.com/project-generator/project_generator_definitions/pull/98
https://github.com/project-generator/project_generator/pull/426

0xc0170, 2016-07-29:
progenef v0.2.28 and progen v0.9.9 should fix this issue, please test

0xc0170, 2016-07-29:
@bcostm @adustm We should review flash algo settings for the nucleo devices, some are not correct for uvision4 projects.

MarceloSalazar, 2016-07-30:
I confirm this works OOB! Awesome work @0xc0170 !

closed 2016-07-30 by MarceloSalazar