ARMmbed/mbed-os #2262: Compiled library includes symbols of mbed-os

jenia81, 2016-07-27:
This issue is copy of https://github.com/ARMmbed/mbed-os/issues/361 issue

When compiling library using:
"mbed compile --library",
in order to add mbed os header files to the include path, "--source" switch is used with the directories of mbed-os.
Those directories contains source files and header files.
The result is that the compiled library include symbols of mbed-os.
What we want, upon library compilation, is that mbed-os symbols will be undefined and they will be added during application compilation

@trianglee 
@mottigondabi
@screamerbg 

Internal Jira was opened in this issue:
ARM Internal Ref: IOTSFW-2686

sg-, 2016-09-22:
Wont fix. Please export and use a makefile.

closed 2016-09-22 by sg-

jenia81, 2016-10-05:
@sg- was there some decision about this issue?
Last mail I saw about it was Doron's mail that says it should be discussed in wider forum

sg-, 2016-10-05:
Best to export to make file using ``` make_arm, make_gcc_arm or make_iar``` and locally modify

jenia81, 2016-10-06:
@sg- As I've wrote in "Dependencies on mbedOS" thread, the maintenance of such makefile seems to be not scalable. I believe that the format of your makefile can change from version to version, so how can we maintain such file? It means that every mbed-os version we need to manually patch the makefile
Unless I'm missing something.

sg-, 2016-10-06:
That would be the suggestion - to export each time a custom build is needed and apply local changes to the makefile