ARMmbed/mbed-os #2263: [exporter] Keil/MDK project compile error for nR51F-DK

toyowata, 2016-07-27:
Tested blinky and used mbed CLI to generate a Keil project using ```mbed export -i uvision5 -m NRF51_DK```.

I installed following packs:
```
* NordicSemiconductor::nRF_DeviceFamilyPack
* NordicSemiconductor::nRF_SoftDevice_Common
* NordicSemiconductor::nRF_SoftDevice_S110
* NordicSemiconductor::nRF_SoftDevice_S130
```

And then, I got following compile errors for build the project:

```
*** Using Compiler 'V5.06 update 1 (build 61)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Unnamed_Project'
compiling softdevice_handler.c...
"no source": Error:  #5: cannot open source input file "..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.c": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.c: 0 warnings, 1 error
compiling softdevice_handler_appsh.c...
"no source": Error:  #5: cannot open source input file "..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c: 0 warnings, 1 error
compiling hci_mem_pool.c...
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c(34): error:  #5: cannot open source input file "hci_mem_pool_internal.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c: 0 warnings, 1 error
compiling btle.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\btle\btle.cpp(35): error:  #5: cannot open source input file "softdevice_handler.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\btle\btle.cpp: 1 warning, 1 error
compiling nRF5xn.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\nRF5xn.cpp(30): error:  #5: cannot open source input file "softdevice_handler.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\nRF5xn.cpp: 1 warning, 1 error
".\.build\uvision5\Unnamed_Project.axf" - 3 Error(s), 2 Warning(s).
Target not created.
Build Time Elapsed:  00:00:04
```

toyowata, 2016-07-27:
This is related to https://github.com/mbedmicro/mbed/issues/2236 and worked around by 
https://github.com/mbedmicro/mbed/issues/2236#issuecomment-235509645

0xc0170, 2016-07-27:
@toyowata I can't reproduce

```
mbed-cli --version
0.8.9

progen --version
0.9.7
```

Using latest master here:

```
mbed-os-example-blinky (master)
λ mbed update
[mbed] Updating program "mbed-os-example-blinky" to latest revision in the current branch
[mbed] Updating library "mbed-os" to rev #5fea6e69ec1a
```

your command : 
```
mbed export -i uvision5 -m NRF51_DK
Scan: lib
Scan: src
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4

Successful:
  * NRF51_DK::uvision5  .\projectfiles\uvision5_NRF51_DK\Unnamed_Project```

builds for me without errors

toyowata, 2016-07-27:
@0xc0170 This is a problem related to Windows path limitation.  When I used Windows subst command and substitute a folder to drive name, I could not see this problem.  You will see this error if your working directory is in deep level (i.e. long path name).

0xc0170, 2016-07-27:
OK I could reproduce these errors:

```
*** Using Compiler 'V5.06 update 1 (build 61)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Unnamed_Project'
compiling softdevice_handler_appsh.c...
"no source": Error:  #5: cannot open source input file "..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c: 0 warnings, 1 error
compiling hci_mem_pool.c...
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c(34): error:  #5: cannot open source input file "hci_mem_pool_internal.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c: 0 warnings, 1 error
compiling softdevice_handler.c...
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.c: 0 warnings, 1 error
compiling nRF5xn.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\nRF5xn.cpp: 1 warning, 1 error
compiling btle.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\btle\btle.cpp: 1 warning, 1 error
".\.build\uvision5\Unnamed_Project.axf" - 4 Error(s), 2 Warning(s).
Target not created.
Build Time Elapsed:  00:00:03
```

 This is however related to uvision rather than exporters. One thing is paths are long, but a tool (=uvision) should be able to handle that or at least to issue a warning, but not a compilation error.

@jkrech - any details for this error?

jkrech, 2016-07-27:
Could you also provide me the buildlog and the commandline options from the C/C++ tab as well as a via files for that project. The commandline options and the source file are usually handed over to the compiler via a file, hence the 260 characters limit should not apply here. If you check to create a batch file you will see the via files in the output folder the extension is <module>.__i

0xc0170, 2016-07-28:
@jkrech:

here are options from C/C++ tab:

Defines:
```
DEVICE_ERROR_PATTERN=1, NRF51, __MBED__=1, TARGET_LIKE_MBED, TARGET_NRF51822, DEVICE_PORTINOUT=1, __MBED_CMSIS_RTOS_CM, TOOLCHAIN_object, __CMSIS_RTOS, TARGET_CORTEX_M, ARM_MATH_CM0, TARGET_UVISOR_UNSUPPORTED, FEATURE_BLE=1, TARGET_M0, TARGET_MCU_NRF51, DEVICE_SERIAL=1, DEVICE_INTERRUPTIN=1, __CORTEX_M0, DEVICE_I2C=1, DEVICE_PORTOUT=1, TARGET_FF_ARDUINO, DEVICE_PORTIN=1, TARGET_RELEASE, TARGET_NORDIC, TARGET_NRF51_DK, DEVICE_SLEEP=1, MBED_BUILD_TIMESTAMP=1469631128, TARGET_MCU_NRF51822, TARGET_MCU_NORDIC_32K, DEVICE_SPI=1, TOOLCHAIN_ARM_STD, DEVICE_SPISLAVE=1, DEVICE_ANALOGIN=1, DEVICE_PWMOUT=1, TOOLCHAIN_ARM, TARGET_LIKE_CORTEX_M0, TARGET_MCU_NRF51_32K, __ASSERT_MSG
```

Misc:
```
 --restrict --multibyte_chars --apcs=interwork -c -O3 -Otime --brief_diagnostics -D__ASSERT_MSG --no_depend_system_headers --preinclude=mbed_config.h --md --cpu=Cortex-M0 --split_sections --no_rtti --gnu
```

The build log:
```
`Output:

*** Using Compiler 'V5.06 update 1 (build 61)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Unnamed_Project'
compiling hci_mem_pool.c...
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c(34): error:  #5: cannot open source input file "hci_mem_pool_internal.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\libraries\hci\hci_mem_pool.c: 0 warnings, 1 error
compiling softdevice_handler_appsh.c...
"no source": Error:  #5: cannot open source input file "..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler_appsh.c: 0 warnings, 1 error
compiling softdevice_handler.c...
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.c: 0 warnings, 1 error
compiling nRF5xn.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\nRF5xn.cpp: 1 warning, 1 error
compiling btle.cpp...
..\..\mbed-os\features\FEATURE_BLE\ble/deprecate.h(26): warning:  #1-D: last line of file ends without a newline
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\softdevice\common\softdevice_handler\softdevice_handler.h(60): error:  #5: cannot open source input file "ble_stack_handler_types.h": No such file or directory
..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\source\btle\btle.cpp: 1 warning, 1 error
".\.build\uvision4\Unnamed_Project.axf" - 4 Error(s), 2 Warning(s).
Target not created.
Build Time Elapsed:  00:00:03
```

jkrech, 2016-07-28:
I have been trying to reproduce a scenario where the paths exceed the 260 character limit.
It is my understanding that the uvproj file has been generated somehow, because if I try to add a file to a project in uVision using the browse dialog the windows system will provide a compacted short path/filename.

If you take a look at your path above in the windows file explorer I am sure you are getting this compacted folder names as well (click into the path and copy to clipboard)
e.g.:
`..\MyExampleFolder\MDK\Boards_need_to_be_extended_further_much_more_to_exeed_260_characters\Keil_this_path_needs_to_be_even_longer\MCBSTM32F400_make_this_a_very_long_path_to_recreate_the_error_mbed_is_seeing\Blinky_and_needs_to_be_even_longer_and_longer_ever_longer_as_long_as_long_gets\Blinky.c`
will be converted into:
`C:\tmp\MYEXAM~1\MDK\BOARDS~1\KEIL_T~1\MCBSTM~1\Blinky_and_needs_to_be_even_longer_and_longer_ever_longer_as_long_as_long_gets\Blinky.c`

If uVision tries to open a file based on the long path/file specification it will fail. The second one succeeds.

The project generator for the uVision project needs to convert the path information in the condensed format as this is the specification of the uvproj file format.

0xc0170, 2016-07-29:
@jkrech Thank you for help! That would explain it. 

> The project generator for the uVision project needs to convert the path information in the condensed format as this is the specification of the uvproj file format.

yes, we use full paths at the moment. I created an issue for progen: https://github.com/project-generator/project_generator/issues/427

0xc0170, 2016-07-29:
@jkrech I added a file to a project, and it prints this warning: 

[image]

The absolute path is <200 characters in this case. What's the meaning of the warning? 

Seems to me this is not a path limit we are hitting as files are not that deep. shouldn't uvision handle this, or as you said it's windows? I mean if I edit manually uvproj, to add a file that has >250 characters, shouldn't uvision do the conversion ? 

It looks to me we are hitting limit with the cmd line limit. I'll investigate. @toyowata please try to pinpoint if this is juts a path problem or there are more things involved

jkrech, 2016-07-29:
The warning is a defect in uVision that will be fixed in the next release 5.21. The limit in fact is 260 characters (MAX_PATH). Sorry about that. The good thing is that despite the warning the file should have been added.

If you edit uvproj files, you are on your own, sorry.
This is an integrated development environment with a proprietary project format.

0xc0170, 2016-07-29:
I understand about the path limit. But I still think this is a problem somewhere else, as the longest path I could find is around 200 characters for this project. We use lot of include paths and defines, and files, let me illustrate by pasting here a control string.

here's the control string for c/c++:

```
--c99 -c --cpu Cortex-M0 -g -O0 --apcs=interwork -I..\.. -I..\..\img -I..\..\mbed-os -I..\..\mbed-os\docs -I..\..\mbed-os\features -I..\..\mbed-os\features\frameworks -I..\..\mbed-os\features\frameworks\greentea-client -I..\..\mbed-os\features\frameworks\greentea-client\greentea-client -I..\..\mbed-os\features\frameworks\greentea-client\source -I..\..\mbed-os\features\frameworks\unity -I..\..\mbed-os\features\frameworks\unity\source -I..\..\mbed-os\features\frameworks\unity\unity -I..\..\mbed-os\features\frameworks\utest -I..\..\mbed-os\features\frameworks\utest\source -I..\..\mbed-os\features\frameworks\utest\utest -I..\..\mbed-os\features\mbedtls -I..\..\mbed-os\features\mbedtls\importer -I..\..\mbed-os\features\mbedtls\inc -I..\..\mbed-os\features\mbedtls\inc\mbedtls -I..\..\mbed-os\features\mbedtls\src -I..\..\mbed-os\features\net -I..\..\mbed-os\features\net\network-socket -I..\..\mbed-os\features\net\network-socket\DnsQuery -I..\..\mbed-os\hal -I..\..\mbed-os\hal\api -I..\..\mbed-os\hal\common -I..\..\mbed-os\hal\hal -I..\..\mbed-os\hal\hal\storage_abstraction -I..\..\mbed-os\hal\targets -I..\..\mbed-os\hal\targets\cmsis -I..\..\mbed-os\hal\targets\cmsis\TARGET_NORDIC -I..\..\mbed-os\hal\targets\cmsis\TARGET_NORDIC\TARGET_MCU_NRF51822 -I..\..\mbed-os\hal\targets\cmsis\TARGET_NORDIC\TARGET_MCU_NRF51822\TOOLCHAIN_ARM_STD -I..\..\mbed-os\hal\targets\cmsis\TARGET_NORDIC\TARGET_MCU_NRF51822\TOOLCHAIN_ARM_STD\TARGET_MCU_NORDIC_32K -I..\..\mbed-os\hal\targets\hal -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822 -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk\components -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk\components\libraries -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk\components\libraries\crc16 -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk\components\libraries\scheduler -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\nordic_sdk\components\libraries\util -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\s110_nrf51822_8_0_0 -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\Lib\s130_nrf51822_1_0_0 -I..\..\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\TARGET_NRF51_DK -I..\..\mbed-os\rtos -I..\..\mbed-os\rtos\rtos -I..\..\mbed-os\rtos\rtx -I..\..\mbed-os\rtos\rtx\TARGET_CORTEX_M -I..\..\mbed-os\rtos\rtx\TARGET_CORTEX_M\TARGET_M0 -I..\..\mbed-os\rtos\rtx\TARGET_CORTEX_M\TARGET_M0\TOOLCHAIN_ARM -I..\..\projectfiles -I..\..\projectfiles\uvision5_NRF51_DK -I..\..\projectfiles\uvision5_NRF51_DK\build -I..\..\mbed-os\features\FEATURE_BLE -I..\..\mbed-os\features\FEATURE_BLE\ble -I..\..\mbed-os\features\FEATURE_BLE\ble\services -I..\..\mbed-os\features\FEATURE_BLE\source -I..\..\mbed-os\features\FEATURE_BLE\source\services -I..\..\mbed-os\features\FEATURE_BLE\targets -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822 -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\bootloader -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\script -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\ble_radio_notification -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\ble_services -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\ble_services\ble_dfu -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\common -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\device_manager -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\device_manager\config -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\ble\peer_manager -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_sdk\components\device -I..\..\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_MCU_NRF51822\nrf51-sdk\source\nordic_s [... cut ...] -DTARGET_MCU_NORDIC_32K -DDEVICE_SPI="1" -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE="1" -DDEVICE_ANALOGIN="1" -DDEVICE_PWMOUT="1" -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K -D__ASSERT_MSG -o ".\.build\uvision5\*.o" --omf_browse ".\.build\uvision5\*.crf" --depend ".\.build\uvision5\*.d" 
```

it's almost 15k characters. Why does it fail if paths are <200 characters (the longest I found was 210 characters, using relative paths). I can provide the example via email if that would help

bridadan, 2016-07-29:
@0xc0170 The ble files definitely have long paths of 200 characters or more. And if you're in a sub directory, chances are you're eating up the other 60 characters there. We still have this issue in our tools as well.

We could fix this by using Unicode paths in Windows, though this will take a substantial amount of work in the tools. I think the pay off would be worth it though!

jkrech, 2016-08-01:
@0xc0170 If you are interested in understanding the way windows short path format works and when it kicks in, I am not the right one to explain that to you in all detail, however what becomes obvious when you play around with it, is the fact that it does not only start kicking in when the MAX_PATH limit is reached (you can use e.g. dir /x to show the corresponding short file name).
This dates back to the times where filenames had a maximum of 8 characters . 3 characters for file extension.

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-