ARMmbed/mbed-os #2264: PIN reassignment for atmel-rf-driver without yotta

markus-becker-tridonic-com, 2016-07-27:
Re-created from https://github.com/ARMmbed/mbed-os/issues/444:

Description

Type: Question
Priority: Major
Question

How to reassign the SPI pins for the Atmel radio?
When using the Atmel radio on a custom target, we would like to reassign the SPI pins.

Should this be done with an own entry in target.json or with a modified mbed_app.json?

If target.json, how am I supposed to set this? Using macros_add? Would you be able to provide an example? I don't see macros added in out put of mbed compile --config.

If mbed_app.json, how am I supposed to set this? I tried with

    "target_overrides": {
        "*": {
            "atmel.rf.spi.mosi": "PB_15"
        }
    }
which resulted in '[ERROR] Invalid parameter name atmel.rf.spi.mosi' in 'application[*]'.

@mlnx

markus-becker-tridonic-com, 2016-07-27:
So I tried 
```
{
    "config": {
        "atmel-rf": {
                    "spi-mosi": "PB_15",
                    "spi-miso": "PB_14",
                    "spi-sclk": "PB_13",
                    "spi-cs": "PB_12",
                    "spi-rst": "PB_1",
                    "spi-slp": "PB_5",
                    "spi-irq": "PA_11",
                    "i2c-sda": "PB_3",
                    "i2c-scl": "PB_10"
        }
```
in my mbed_app.json. This results in:
```
$ mbed compile --config
Configuration parameters
------------------------
atmel-rf.spi-miso has no value
atmel-rf.spi-slp has no value
atmel-rf.i2c-scl has no value
atmel-rf.spi-sclk has no value
app.atmel-rf has no value
atmel-rf.spi-cs has no value
atmel-rf.i2c-sda has no value
atmel-rf.spi-rst has no value
atmel-rf.spi-mosi has no value
atmel-rf.spi-irq has no value
```

With
```
"target_overrides": {
        "*": {
            "atmel-rf": {
                        "spi-mosi": "PB_15",
                        "spi-miso": "PB_14",
                        "spi-sclk": "PB_13",
                        "spi-cs": "PB_12",
                        "spi-rst": "PB_1",
                        "spi-slp": "PB_5",
                        "spi-irq": "PA_11",
                        "i2c-sda": "PB_3",
                        "i2c-scl": "PB_10"
            }
        }
    }
```
the result is
```
$ mbed compile --config
[ERROR] Attempt to override undefined parameter 'app.atmel-rf' in 'application[*]'
```

0xc0170, 2016-07-27:
cc @kjbracey-arm @SeppoTakalo

markus-becker-tridonic-com, 2016-07-27:
Only changing it in mbed_lib.json of atmel-rf-driver works ATM.

bogdanm, 2016-07-27:
Try this in your `mbed_app.json`:

```
"target_overrides": {
        "*": {
                   "atmel-rf.spi-mosi": "PB_15",
                    "atmel-rf.spi-miso": "PB_14",
                    "atmel-rf.spi-sclk": "PB_13",
                    "atmel-rf.spi-cs": "PB_12",
                    "atmel-rf.spi-rst": "PB_1",
                    "atmel-rf.spi-slp": "PB_5",
                    "atmel-rf.spi-irq": "PA_11",
                    "atmel-rf.i2c-sda": "PB_3",
                    "atmel-rf.i2c-scl": "PB_10"
        }
    }
```

markus-becker-tridonic-com, 2016-07-28:
```
"target_overrides": {
        "*": {
                    "atmel-rf.spi-mosi": "PB_15",
                    "atmel-rf.spi-miso": "PB_14",
                    "atmel-rf.spi-sclk": "PB_13",
                    "atmel-rf.spi-cs": "PB_12",
                    "atmel-rf.spi-rst": "PB_1",
                    "atmel-rf.spi-slp": "PB_5",
                    "atmel-rf.spi-irq": "PA_11",
                    "atmel-rf.i2c-sda": "PB_3",
                    "atmel-rf.i2c-scl": "PB_10"
        }
    }
```

This works. Thanks.

closed 2016-07-28 by markus-becker-tridonic-com