ARMmbed/mbed-os #2265: Cannot enable feature COMMON_PAL

SeppoTakalo, 2016-07-27:
I cannot enable the feature **COMMON_PAL** required by Client or IPv6.

In my `mbed_app.json` I have:
```
"target.features_add": ["COMMON_PAL", "IPV6"],
```

Result is:

```
mbed-os-example-mesh-minimal$ mbed compile
Building project mbed-os-example-mesh-minimal (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[ERROR] Feature 'COMMON_PAL' is not a supported features
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/septak01/src/mbed-os-example-mesh-minimal/.temp/tools/make.py -t GCC_ARM -m K64F --source . --build ./.build/K64F/GCC_ARM" in "/Users/septak01/src/mbed-os-example-mesh-minimal"
---
```

Example application tried was: https://github.com/ARMmbed/mbed-os-example-mesh-minimal
with mbed-os replaced with `mbedmicro/mbed`

SeppoTakalo, 2016-07-27:
CC: @sg- @screamerbg @MarceloSalazar @0xc0170

sg-, 2016-07-27:
Puzzled. This should be possible https://github.com/mbedmicro/mbed/blob/1e069aa00963a158341ea0554499d65275e73ec9/tools/config.py#L189

SeppoTakalo, 2016-07-27:
Found a probable cause:
```
$ mbed compile -c
[mbed] ERROR: The mbed tools were not found in "/Users/septak01/src/mbed-os-example-mesh-minimal". 
[mbed] ERROR:  Run `mbed deploy` to install dependencies and tools. 
---
```

After running a mbed deploy, it download some other toolset than what is in `mbed/tools`
```
$ mbed deploy
[mbed] Updating library "atmel-rf-driver" to rev #a16a4ed6f6e8
[mbed] Updating library "mbed" to rev #b171a9c4b853
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

```
 find . -name config.py
./.temp/tools/config.py
./mbed/tools/config.py
```

The one in `mbed/tools/config.py` allows `COMMON_PAL` but the one in .temp does not.

SeppoTakalo, 2016-07-27:
Maybe the problem is that why it does not find those build tools.

SeppoTakalo, 2016-07-27:
After renaming `mbed` to `mbed-os` the problem disappears, so I'm assuming **mbed-cli** assumes to find mbed-os from that path, and not from `mbed`

This should probably be documented. Just `mbed add https://github.com/mbedmicro/mbed` is not enough.

sg-, 2016-07-27:
Good catch! @screamerbg

0xc0170, 2016-07-27:
Reference: https://github.com/ARMmbed/mbed-cli/blob/485cbc6da77b7dbe68a5bf9aa105a407d40da7ea/mbed/mbed.py#L1155

This shall be reported to mbed-cli, and fixed there.

sg-, 2016-07-27:
Well the repo will be named mbed-os. Once tag for rc2 is there core.lib should be updated and mbed-os can be used

0xc0170, 2016-07-27:
> Well the repo will be named mbed-os. Once tag for rc2 is there core.lib should be updated and mbed-os can be used

Yes, that's correct, I was referring to this:

> This should probably be documented. Just mbed add https://github.com/mbedmicro/mbed is not enough.

That if it's not, and should be, it should be tracked in mbed-cli, not here.

@SeppoTakalo happy to close this, and request a doc update on mbed-cli?

SeppoTakalo, 2016-07-27:
Opened under mbed-cli as a documentation request: https://github.com/ARMmbed/mbed-cli/issues/296

closed 2016-07-27 by SeppoTakalo