ARMmbed/mbed-os #2267: Build does not respect dependency when triggered from different directory level

MarceloSalazar, 2016-07-27:
### Description
- Type: Bug
- Priority:  Major

Issue reported during SiP workshop.

Need to check the actual behaviour and fix potential issues

ARM Internal Ref: IOTSFW-2732

Originally raised here: https://github.com/ARMmbed/mbed-os/issues/39

0xc0170, 2016-07-27:
> Originally raised here: ARMmbed/mbed-os#39

39 does not look like the right reference to this issue?

How to reproduce? What's the expected behavior? Can you add more details

MarceloSalazar, 2016-07-27:
sorry, I missed a '0'
https://github.com/ARMmbed/mbed-os/issues/390

sg-, 2016-09-22:
@screamerbg can you respond about this and if it will be fixed. I recall something about the CWD and ```../../../``` being appended to the .d files or something similar

theotherjimmy, 2017-04-25:
I cannot reproduce this issue. @MarceloSalazar Has this been solved in the mean time?

MarceloSalazar, 2017-04-28:
I cannot reproduce this anymore.

closed 2017-04-28 by MarceloSalazar