ARMmbed/mbed-os #2268: mbed-os-example-uvisor GCC_ARM export doesn't work on both Linux and Mac for mbed-os-5.1.0-rc1

bulislaw, 2016-07-27:
# List of steps
1) mbed import mbed-os-exampkle-uvisor
2) mbed update mbed-os-5.1.0-rc1
3) mbed export -i GCC_ARM -m K64F
4) cd projectfiles/gcc_arm_K64F/
5) cp .build/Unnamed_Project.bin /Volumes/MBED/

# Results
LED stays off. It works fine for "mbed compile -m K64F -t GCC_ARM"

# Symbol differences
## mbed compile
```
+-------------------------+-------+-------+------+
| Module                  | .text | .data | .bss |
+-------------------------+-------+-------+------+
| Fill                    |   149 |   508 | 2324 |
| Misc                    | 52341 |  1696 | 3873 |
| features/FEATURE_UVISOR |   646 |     0 |    1 |
| features/frameworks     |  3288 |    52 |  328 |
| features/net            |    12 |     0 |    0 |
| hal/common              |  2309 |     0 |  269 |
| hal/targets             | 12787 |    12 |  200 |
| rtos/rtos               |   258 |     4 |    0 |
| rtos/rtx                |  5775 |    20 | 2685 |
| Subtotals               | 77565 |  2292 | 9680 |
+-------------------------+-------+-------+------+
Allocated Heap: 65536 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 11972 bytes
Total RAM memory (data + bss + heap + stack): 110276 bytes
Total Flash memory (text + data + misc): 80897 bytes
```

## make
```
  text	   data	    bss	    dec	    hex	filename
69312	   2348	 251740	 323400	  4ef48	Unnamed_Project.elf
```

# Compilation status
I'm getting warning from the make:
```
/usr/local/Caskroom/gcc-arm-embedded/5_4-2016q2,20160622/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: alignment 1 of symbol `__uvisor_ps' in ../../../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(uvisor-output.o) is smaller than 4 in ../../../mbed-os/features/FEATURE_UVISOR/targets/TARGET_UVISOR_SUPPORTED/TARGET_MCU_K64F/TARGET_RELEASE/TARGET_M4/libconfiguration_kinetis_m4_0x1fff0000.a(disabled.o)
```

It seems that the all .test .data and .bss generated after export have different size.

0xc0170, 2016-07-27:
I am not certain if this is not supposed to go to uvisor example.

cc @AlessandroA

bulislaw, 2016-07-27:
Moved to https://github.com/ARMmbed/mbed-os-example-uvisor/issues/8

closed 2016-07-27 by bulislaw