ARMmbed/mbed-os #2270: [exporter] mbed-os-5.1.0-rc1: export to GCC_ARM broken for nRF51_dk

bulislaw, 2016-07-27:
Binaries generated after export to GCC_ARM by Makefile don't work for multiple demos.

# OS
MacOS (10.11.6)
Ubuntu 14.04

# Tag
mbed-os-5.1.0-rc1

# Log
## Makefile:
```
arm-none-eabi-objcopy -O binary Unnamed_Project.elf Unnamed_Project.bin
arm-none-eabi-size Unnamed_Project.elf
   text	   data	    bss	    dec	    hex	filename
  60180	    264	   3168	  63612	   f87c	Unnamed_Project.elf
```
```
 $ ls -lh .build/
total 800
-rwxr-xr-x  1 barsza01  staff    42K 27 Jul 10:53 Unnamed_Project.bin
-rwxr-xr-x  1 barsza01  staff   219K 27 Jul 10:53 Unnamed_Project.elf
-rw-r--r--  1 barsza01  staff   119K 27 Jul 10:53 Unnamed_Project.hex
-rw-r--r--  1 barsza01  staff   8.8K 27 Jul 10:52 main.d
-rw-r--r--  1 barsza01  staff   3.4K 27 Jul 10:52 main.o
drwxr-xr-x  5 barsza01  staff   170B 27 Jul 10:52 mbed-os
```

## mbed

```
+----------------------+-------+-------+------+
| Module               | .text | .data | .bss |
+----------------------+-------+-------+------+
| Fill                 |   127 |     3 |   29 |
| Misc                 |  8152 |   120 |  500 |
| features/FEATURE_BLE | 21670 |    13 |  835 |
| features/frameworks  |  3142 |    92 |  336 |
| features/net         |    16 |     0 |    0 |
| hal/common           |  2472 |     4 |  269 |
| hal/targets          |  2794 |     8 |  209 |
| rtos/rtos            |   442 |     4 |    0 |
| rtos/rtx             |  5137 |    20 |  990 |
| Subtotals            | 43952 |   264 | 3168 |
+----------------------+-------+-------+------+
Allocated Heap: 17048 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 3432 bytes
Total RAM memory (data + bss + heap + stack): 22528 bytes
Total Flash memory (text + data + misc): 44216 bytes
```
```
$ ls -lh .build/NRF51_DK/GCC_ARM/
total 3344
-rw-r--r--  1 barsza01  staff   3.6K 27 Jul 10:09 main.d
-rw-r--r--  1 barsza01  staff   3.4K 27 Jul 10:09 main.o
drwxr-xr-x  5 barsza01  staff   170B 27 Jul 10:09 mbed-os
-rwxr-xr-x  1 barsza01  staff   219K 27 Jul 10:09 mbed-os-example-blinky.elf
-rwxr-xr-x  1 barsza01  staff   413K 27 Jul 10:14 mbed-os-example-blinky.hex
-rw-r--r--  1 barsza01  staff   1.0M 27 Jul 10:09 mbed-os-example-blinky.map
-rw-r--r--  1 barsza01  staff   614B 27 Jul 10:14 mbed-os-example-blinky_map.csv
-rw-r--r--  1 barsza01  staff   1.5K 27 Jul 10:14 mbed-os-example-blinky_map.json
```

Especially the .hex file is 413K vs 119K

0xc0170, 2016-07-27:
cc @pan-

pan-, 2016-07-27:
Looks like the softdevice is not merged in the final binary.

0xc0170, 2016-07-27:
@pan- is looking at it (somehow I could not assign this to him, we shall add him to the team !)

bogdanm, 2016-08-02:
@pan-: any update on this?

closed 2016-08-03 by sg-