ARMmbed/mbed-os #2271: [exporter] Keil/MDK project does not have correct flash algorithm for NUCLEO_F429ZI

MarceloSalazar, 2016-07-27:
Tested blinky and used mbed CLI to generate a Keil project.
Attempted to download code to NUCLEO_F429ZI target and found out doesn't work OOB.
I needed to remove the flash algorithm and install it again.

@bcostm @adustm @ohagendorf @0xc0170

0xc0170, 2016-07-27:
I believe this is connected to 2254. Will be fixed via a fix to progen. I'll reference it here once done

0xc0170, 2016-07-27:
This one adds uvopt for uvision4: https://github.com/project-generator/project_generator/pull/426

the another issue might be wrong definitons for this target. Please set the correct flash algo, use
```
progendef create -f .uvproj_file_path -t uvision -m f429
```

that produces a yaml file, please compare that one to the https://github.com/project-generator/project_generator_definitions/blob/master/project_generator_definitions/mcu/st/stm32f429zi.yaml#L25

I believe it needs a correction

0xc0170, 2016-07-27:
This should follow f411 fix, once that one gets fixed, this shall be retested.

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-