ARMmbed/mbed-os #2272: [exporter] [GCC_ARM] The build command for ASM files does not include custom defines

AlessandroA, 2016-07-27:
If you generate an IDE project file for `GCC_ARM` (a `Makefile`), the build command for assembly files does not include macro definitions:

With `mbed compile`:

```text
[DEBUG] Compile: arm-none-eabi-gcc -x assembler-with-cpp -c -Wall -Wextra 
-Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 
-fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char 
-MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m4 
-mthumb -Os -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED 
-DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 
-DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED 
-DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT 
-DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DTARGET_CORTEX_M 
-DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 
-DTARGET_M4 -DTARGET_K64F -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 
-DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 
-DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 
-DMBED_BUILD_TIMESTAMP=1469628102.58 -DTARGET_UVISOR_SUPPORTED 
-DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE 
-DDEVICE_SERIAL_FC=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 
-DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DDEVICE_SPI=1 
-DFEATURE_UVISOR=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 
-DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F 
-DARM_MATH_CM4 @./.build/K64F/GCC_ARM/.includes_e0afd731fb329ca7624744d1d649d0a2.txt
-include ./.build/K64F/GCC_ARM/mbed_config.h
-o ./.build/K64F/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.o
./mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.S
```

With `mbed export` followed by `make`:

```text
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp 
-c -x assembler-with-cpp -c -Wall -Wextra -Wno-unused-parameter
-Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin
-ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks
-fomit-frame-pointer -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp
-Os -o mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.o ../../../mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.S
```

(verified with the uVisor example; see [this issue](https://github.com/ARMmbed/mbed-os-example-uvisor/issues/8) to reproduce it).

@geky @sg- @bogdanm @theotherjimmy @0xc0170

0xc0170, 2016-07-28:
@bogdanm @theotherjimmy Can you have a look?

My quick guess would be that -D for make might be a problematic for windows, but there should be --preinclude. I could not spot it in the pasted cmd above.

theotherjimmy, 2016-07-28:
@0xc0170 you are exactly correct. #2303 should fix it.

PR #2303 (2016-08-02): Add missing --preinclude and -input on assembly files
Except for IAR

Resolves #2272
closed 2016-08-02 by bogdanm