ARMmbed/mbed-os #2275: invalid warning message about project-generator-definitions not installed

maclobdell, 2016-07-27:
I get a warning when importing the examples that project-generator-definitions is missing.   However, I had it installed (it was a local version of project-generator-definitions installed in developer mode).  I also tried installing all the requirements per the instructions, but I get the same warning message that it is not installed.

```
C:\Users\maclob01\Documents\Projects\OOB_16_07>mbed import mbed-os-example-uvisor
[mbed] Importing program "mbed-os-example-uvisor" from "https://github.com/ARMmbed/mbed-os-example-uvisor" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/mbedmicro/mbed" at rev #5fea6e69ec1a
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: project_generator_definitions
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor\mbed-os"
---

C:\Users\maclob01\Documents\Projects\OOB_16_07>cd mbed-os-example-uvisor

C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor>mbed update mbed-os-5.1.0-rc1
[mbed] Updating program "mbed-os-example-uvisor" to branch mbed-os-5.1.0-rc1
[mbed] Updating library "mbed-os" to rev #5fea6e69ec1a
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: project_generator_definitions
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor\mbed-os"
---

C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor>cd mbed-os

C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor\mbed-os>pip install -r requirements.txt
Requirement already satisfied (use --upgrade to upgrade): colorama>=0.3.3 in c:\python27\lib\site-packages (from -r requirements.txt (line 1))
Requirement already satisfied (use --upgrade to upgrade): PySerial>=2.7 in c:\python27\lib\site-packages (from -r requirements.txt (line 2))
Requirement already satisfied (use --upgrade to upgrade): PrettyTable>=0.7.2 in c:\python27\lib\site-packages (from -r requirements.txt (line 3))
Requirement already satisfied (use --upgrade to upgrade): Jinja2>=2.7.3 in c:\python27\lib\site-packages (from -r requirements.txt (line 4))
Requirement already satisfied (use --upgrade to upgrade): IntelHex>=1.3 in c:\python27\lib\site-packages (from -r requirements.txt (line 5))
Collecting project-generator<0.10.0,>=0.9.7 (from -r requirements.txt (line 6))
Collecting project_generator_definitions<0.3.0,>=0.2.26 (from -r requirements.txt (line 7))
Requirement already satisfied (use --upgrade to upgrade): junit-xml in c:\python27\lib\site-packages (from -r requirements.txt (line 8))
Requirement already satisfied (use --upgrade to upgrade): pyYAML in c:\python27\lib\site-packages (from -r requirements.txt (line 9))
Requirement already satisfied (use --upgrade to upgrade): requests in c:\python27\lib\site-packages (from -r requirements.txt (line 10))
Requirement already satisfied (use --upgrade to upgrade): mbed-ls>=0.2.13 in c:\python27\lib\site-packages (from -r requirements.txt (line 11))
Requirement already satisfied (use --upgrade to upgrade): mbed-host-tests>=0.2.18 in c:\python27\lib\site-packages (from -r requirements.txt (line 12))
Requirement already satisfied (use --upgrade to upgrade): mbed-greentea>=0.2.24 in c:\python27\lib\site-packages (from -r requirements.txt (line 13))
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in c:\python27\lib\site-packages (from Jinja2>=2.7.3->-r requirements.txt (line 4))
Requirement already satisfied (use --upgrade to upgrade): argparse in c:\python27\lib\site-packages (from project-generator<0.10.0,>=0.9.7->-r requirements.txt (line 6))
Requirement already satisfied (use --upgrade to upgrade): xmltodict in c:\python27\lib\site-packages (from project-generator<0.10.0,>=0.9.7->-r requirements.txt (line 6))
Requirement already satisfied (use --upgrade to upgrade): six in c:\python27\lib\site-packages (from junit-xml->-r requirements.txt (line 8))
Requirement already satisfied (use --upgrade to upgrade): lockfile in c:\python27\lib\site-packages (from mbed-ls>=0.2.13->-r requirements.txt (line 11))
Requirement already satisfied (use --upgrade to upgrade): mock in c:\python27\lib\site-packages (from mbed-greentea>=0.2.24->-r requirements.txt (line 13))
Requirement already satisfied (use --upgrade to upgrade): funcsigs>=1; python_version < "3.3" in c:\python27\lib\site-packages (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 13))
Requirement already satisfied (use --upgrade to upgrade): pbr>=0.11 in c:\python27\lib\site-packages (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 13))
Installing collected packages: project-generator-definitions, project-generator
  Found existing installation: project-generator-definitions 0.2.25
    Uninstalling project-generator-definitions-0.2.25:
      Successfully uninstalled project-generator-definitions-0.2.25
  Found existing installation: project-generator 0.8.16
    Uninstalling project-generator-0.8.16:
      Successfully uninstalled project-generator-0.8.16
Successfully installed project-generator-0.9.8 project-generator-definitions-0.2.25

C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor\mbed-os>cd ..

C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor>mbed update mbed-os-5.1.0-rc1
[mbed] Updating program "mbed-os-example-uvisor" to branch mbed-os-5.1.0-rc1
[mbed] Updating library "mbed-os" to rev #5fea6e69ec1a
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed build tools in this program require Python modules that are not installed.
[mbed] WARNING: This might prevent compiling code or exporting to IDEs and other toolchains.
[mbed] WARNING: The missing Python modules are: project_generator_definitions
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\maclob01\Documents\Projects\OOB_16_07\mbed-os-example-uvisor\mbed-os"
---
```

sg-, 2016-07-27:
@0xc0170

0xc0170, 2016-07-28:
I'll have a look. This might be a bug in the requirements or how mbed-cli checks those.

```
project-generator>=0.9.7,<0.10.0
project_generator_definitions>=0.2.26,<0.3.0
```

It should update to at least to v0.2.26.  The current version in pypi is 0.2.27 (https://pypi.python.org/pypi/project_generator_definitions)

cc @screamerbg

0xc0170, 2016-07-28:
I'll send a fix, this is a problem of pip and python module naming (dashes vs underscores). The req file in mbed uses ``_`` for definitions, and that causes a problem for mbed-cli ``get_requirements`` function

0xc0170, 2016-07-28:
I sent a fix, waiting for ppl to review, referenced above

0xc0170, 2016-07-29:
Should be fixed on master now, please test

PR #2284 (2016-07-29): Requirements - fix progen def name - use dashes
Fixes #2275. Reference:
http://pybuilder.github.io/documentation/external_plugins.html#.V5m-hfmANBc
commit 9b89fc3b (2016-07-29): Requirements - fix progen def name - use dashes

Fixes #2275. Reference:
http://pybuilder.github.io/documentation/external_plugins.html#.V5m-hfmANBc
closed 2016-07-29 by 0xc0170