ARMmbed/mbed-os #2276: Inconsistent thread safety implementation

neilt6, 2016-07-27:
Most of the new thread safe classes are using a single static mutex for all instances. This guarantees thread safety across multiple objects, but prevents multiple independent peripherals from being used simultaneously. Kind of inconvenient, but developers can override the virtual `lock()` and `unlock()` methods to disable the mutex so not a big deal.

However, `AnalogOut`, `CAN`, `SPI`, and `Serial` are using _separate_ mutexes for each instance. This guarantees nothing unless the developer creates a single `SPI` object and passes it around, which is not the way these objects have historically been used. Is this implementation a mistake, or by design?

0xc0170, 2016-07-27:
cc @c1728p9

c1728p9, 2016-07-27:
Hi @neilt6, this behavior was intentional for AnalogOut, CAN, and Serial since the typical use case it to only have one instance of these objects per hardware resource. SPI, similar to IIC, can have multiple instances bound to the same hardware, so this should be using a shared mutex.

neilt6, 2016-07-27:
@c1728p9 - Thanks for the clarification. My main concern was `SPI`, I'm less familiar with the others. Would you like me to submit a pull-request modifying `SPI` to use a `SingletonPtr`?

c1728p9, 2016-07-27:
@neilt6, yeah a PR would be great! Thanks

closed 2016-07-29 by neilt6