ARMmbed/mbed-os #2283: [NRF51_DK] us_ticker.c doesn't support IAR build

toyowata, 2016-07-28:
The RTC1_IRQHandler in us_ticker.c is implemented by inline assembler and there is no IAR toolchain support.  Detected by RC1 OOB test.

```
C:\Users\toywat01\Documents\GitHub\mbed-os-example-blinky>mbed compile -m NRF51_DK -t IAR
Building project mbed-os-example-blinky (NRF51_DK, IAR)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: mbed
Scan: env
Compile: rt_MemBox.c
Compile: rt_Memory.c
Compile: us_ticker.c
[ERROR]
  #error Compiler not supported.
   ^
"C:\Users\toywat01\Documents\GitHub\mbed-os-example-blinky\mbed-os\hal\targets\hal\TARGET_NORDIC\TARGET_MCU_NRF51822\us_ticker.c",382  Fatal error[Pe035]: #error directive: Compiler not supported.
Fatal error detected, aborting.

```

https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/us_ticker.c#L382

0xc0170, 2016-07-28:
cc @pan-

pan-, 2016-07-28:
@toyowata This is expected for the RC1 because IAR support was not enabled.
It will be for RC2.

toyowata, 2016-07-28:
@pan- Thanks. I will close this issue, once I confirm the fix in RC2.

toyowata, 2016-07-29:
@pan- Confirmed the fix.  Closed ticket.

closed 2016-07-29 by toyowata