ARMmbed/mbed-os #2286: NUCLEO_F429ZI tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: NUCLEO_F429ZI
Host OS: Linux Ubuntu 14.04

```

+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.69               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.69               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 64.61              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.55              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.63              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.57               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.13               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.22               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.32               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.48               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.28              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.34               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.54               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.51               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-integration-basic                                                      | OK      | 7.86               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-integration-threaded_blinky                                            | OK      | 27.85              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 10.38              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-callback                                                  | TIMEOUT | 64.69              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-dev_null                                                  | TIMEOUT | 64.71              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 20.05              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-generic_tests                                             | TIMEOUT | 66.72              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-rtc                                                       | TIMEOUT | 66.79              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-stl_features                                              | TIMEOUT | 66.76              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-ticker                                                    | TIMEOUT | 64.6               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 19.46              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.32              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK      | 10.42              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.03               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.19               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-mbed-div                                                     | OK      | 8.17               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 22.82              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 13.14              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | TIMEOUT | 64.63              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.26              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT | 65.63              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.43              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | OK      | 13.14              | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | OK      | 13.7               | shell       |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 22.85              | shell       |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 30 OK / 9 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--                ------+--------------------+
| target                | platform_name | test suite                                                                           | test case                                                              | passed | failed | r                esult | elapsed_time (sec) |
+-----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--                ------+--------------------+
[... cut ...]
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                                | 1      | 0      | O                K     | 22.85              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--                ------+--------------------+
mbedgt: test case results: 93 OK / 9 ERROR
mbedgt: completed in 973.42 sec
```

screamerbg, 2016-07-28:
CC @adustm

bridadan, 2016-07-28:
There is an issue with ST-Link on Linux where the disk drive appears to "fill up" after multiple binaries have been flashed to the board. This causes flashing to fail with the testing tools and requires the board to be manually unplugged then plugged back into the machine. @adustm we have raised this issue before, have you heard anything about this issue recently?

adustm, 2016-07-29:
Hello, have you tried to update the stlink FW with the last version ? If the issue remains, do not hesitate to contact myst.com forums

c1728p9, 2016-08-02:
I can confirm that this runs on windows.

c1728p9, 2016-08-02:
[test_f429_68d48ea5412be1f94a1e67afb7d008d80048abc7.txt](https://github.com/ARMmbed/mbed-os/files/397383/test_f429_68d48ea5412be1f94a1e67afb7d008d80048abc7.txt)

Windows test results attached

bridadan, 2016-08-03:
Thanks @c1728p9, I'm fine with closing this issue then since this seems to be an issue between the interface firmware and the OS.

closed 2016-08-04 by sg-