ARMmbed/mbed-os #2287: NUCLEO_F01RE tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: NUCLEO_F401RE
Host OS: Windows 7

```

+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.85               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.85               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 10.49              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.69              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.79              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.7                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.3                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.39               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.44               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.62               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.3               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.56               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.67               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.67               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-basic                                                      | OK      | 8.05               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-threaded_blinky                                            | OK      | 27.98              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 10.52              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-callback                                                  | OK      | 14.9               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-dev_null                                                  | OK      | 10.94              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 20.33              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-generic_tests                                             | OK      | 9.2                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-rtc                                                       | OK      | 28.81              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-stl_features                                              | OK      | 9.82               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-ticker                                                    | OK      | 39.9               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 19.66              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.53              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK      | 10.55              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.18               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.32               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-div                                                     | OK      | 8.34               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 23.0               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 13.25              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | OK      | 9.7                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.37              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | OK      | 9.72               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.59              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | OK      | 13.27              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 48.12              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 23.0               | shell       |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 38 OK / 1 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+---------+--------------------+
| target                | platform_name | test suite                                                                           | test case                                              | passed | faile
d | result  | elapsed_time (sec) |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+---------+--------------------+
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                | 1      | 0
  | OK      | 23.0               |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+---------+--------------------+
mbedgt: test case results: 13 SKIPPED / 119 OK / 2 ERROR
mbedgt: completed in 566.91 sec

```

screamerbg, 2016-07-28:
CC @adustm

MarceloSalazar, 2016-07-30:
RC#4

```
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.84               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.9                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 10.58              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.71              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.84              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.72               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.27               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.33               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.44               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.59               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.31              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.5                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.65               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.69               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-basic                                                      | OK      | 8.03               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-threaded_blinky                                            | OK      | 28.02              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 10.53              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-callback                                                  | OK      | 14.89              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-dev_null                                                  | OK      | 10.91              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 20.49              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-generic_tests                                             | OK      | 9.25               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-rtc                                                       | OK      | 20.03              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-stl_features                                              | OK      | 9.87               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-ticker                                                    | OK      | 39.95              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 19.65              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.53              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK      | 10.54              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.19               | shell       |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                | 1      | 0
  | OK      | 23.01              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+---------+--------------------+
mbedgt: test case results: 13 SKIPPED / 120 OK / 1 ERROR
mbedgt: completed in 558.59 sec
```

adustm, 2016-11-22:
Hello,
I just checked in our test history.  This test is ok at least since mbed-os-5.1.4
Here is the output for NUCLEO_F401RE / GCC_ARM / mbed-os-tests-mbedmicro-rtos-mbed-threads on today's mbed-os database (sha1 = f4007e296d235bdf8b0d402efeebebbb3b261951)
```
+-----------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                                                                   | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | OK     | 18.76              | shell       |
```
```
+-----------------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                                                                   | test case                                              | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing parallel threads                               | 1     | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing parallel threads with child                    | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing parallel threads with murder                   | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing parallel threads with wait                     | 1     | 0      | OK     | 0.16               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing parallel threads with yield                    | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing serial threads                                 | 1     | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing serial threads with child                      | 1     | 0      | OK     | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing serial threads with murder                     | 1     | 0      | OK     | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing serial threads with wait                       | 1     | 0      | OK     | 1.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing serial threads with yield                      | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing single thread                                  | 1     | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing single thread with child                       | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing single thread with murder                      | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing single thread with wait                        | 1     | 0      | OK     | 0.15               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing single thread with yield                       | 1     | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads                                            | Testing thread self terminate                          | 1     | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-timer                                              | tests-mbedmicro-rtos-mbed-timer                        | 1     | 0      | OK     | 23.41              |
```

Could you tell us if you still have the issue ? we don't reproduce it.

Kind regards
Armelle

closed 2017-01-16 by sg-