ARMmbed/mbed-os #2288: DISCO_F429ZI tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: DISCO_F429ZI
Host OS: Linux Ubuntu 14.04

```

+----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target               | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.68               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.68               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 64.61              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.57              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.61              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.53               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.1                | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.21               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.27               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.44               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 12.27              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.32               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.47               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.48               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-integration-basic                                                      | OK      | 7.86               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-integration-threaded_blinky                                            | OK      | 27.78              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 10.37              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-callback                                                  | TIMEOUT | 64.69              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-dev_null                                                  | TIMEOUT | 64.71              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 19.99              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-generic_tests                                             | TIMEOUT | 66.7               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-rtc                                                       | TIMEOUT | 66.75              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-stl_features                                              | TIMEOUT | 66.74              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-ticker                                                    | TIMEOUT | 64.61              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 19.46              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.29              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK      | 10.37              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.02               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.14               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-mbed-div                                                     | OK      | 8.12               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 22.81              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 13.06              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | TIMEOUT | 64.68              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.25              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT | 65.54              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.42              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | OK      | 13.15              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | OK      | 13.68              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 22.86              | shell       |
+----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 30 OK / 9 TIMEOUT
mbedgt: test case report:
+----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target               | platform_name | test suite                                                                           | test case                                                              | passed | failed | result | elapsed_time (sec) |
+----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
[... cut ...]
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                                | 1      | 0      | OK     | 22.86              |
+----------------------+---------------+--------------------------------------------------------------------------------------+------------------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 93 OK / 9 ERROR
mbedgt: completed in 974.53 sec

```

screamerbg, 2016-07-28:
CC @adustm

bridadan, 2016-07-28:
I mentioned this in #2286, but I'll go ahead and copy it below:

> There is an issue with ST-Link on Linux where the disk drive appears to "fill up" after multiple binaries have been flashed to the board. This causes flashing to fail with the testing tools and requires the board to be manually unplugged then plugged back into the machine. @adustm we have raised this issue before, have you heard anything about this issue recently?

bulislaw, 2016-08-03:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc5
```
mbedgt: test suite report:
+----------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target               | platform_name | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+----------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 9.25               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.23               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 11.09              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 17.12              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 11.67              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.26               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.87               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.86               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT | 67.08              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 10.5               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 11.59              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 9.01               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 9.16               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 9.33               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-integration-basic                                                      | OK      | 8.9                | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-integration-threaded_blinky                                            | OK      | 28.98              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-c_strings                                                 | OK      | 11.47              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-callback                                                  | TIMEOUT | 66.63              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-dev_null                                                  | TIMEOUT | 67.57              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-echo                                                      | OK      | 21.0               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-generic_tests                                             | TIMEOUT | 67.06              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-rtc                                                       | OK      | 20.24              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-stl_features                                              | OK      | 10.44              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-ticker                                                    | OK      | 40.65              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-timeout                                                   | OK      | 20.44              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbed_drivers-wait_us                                                   | OK      | 19.17              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbedmicro-mbed-attributes                                              | OK      | 10.97              | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbedmicro-mbed-call_before_main                                        | OK      | 9.03               | shell       |
| DISCO_F429ZI-GCC_ARM | DISCO_F429ZI  | tests-mbedmicro-mbed-cpp                                                     | TIMEOUT | 67.1               | shell       |
[... cut ...]
+----------------------+---------------+------------------------------------------------------------------------------+---------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 101 OK / 8 ERROR
mbedgt: completed in 989.83 sec
mbedgt: exited with code 8
[mbed] ERROR: "mbedgt" returned error code 8.
[mbed] ERROR: Command "mbedgt --test-spec /Users/barsza01/devel/mbed/mbed-os/.build/tests/DISCO_F429ZI/GCC_ARM/test_spec.json" in "/Users/barsza01/devel/mbed/mbed-os"
---
```

adustm, 2016-11-22:
Hello,
The status on mbed-os-5.2.2 is the following on ST side :
```
+-----------------------+---------------+------------------------------------------------------------------------------+----------------------------------------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                                                                   | test case                                                            | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+------------------------------------------------------------------------------+----------------------------------------------------------------------+--------+--------+--------+--------------------+
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                                       | 2      | 0      | OK     | 0.08               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                                          | 1      | 0      | OK     | 0.03               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test_default                | Repeating Test                                                       | 2      | 0      | OK     | 0.08               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test_default                | Simple Test                                                          | 1      | 0      | OK     | 0.03               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Cancel Repeat                       | 1      | 0      | OK     | 0.18               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Enable Repeat Handler               | 2      | 0      | OK     | 0.19               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Premature Validation                              | 1      | 0      | OK     | 0.07               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Validation                                        | 1      | 0      | OK     | 0.16               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Premature Validation                                       | 1      | 0      | OK     | 0.06               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Simple Validation                                          | 1      | 0      | OK     | 0.06               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Await                                                       | 1      | 0      | OK     | 1.42               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: CaseNext                                                    | 1      | 0      | OK     | 0.04               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: NoTimeout                                                   | 1      | 0      | OK     | 0.05               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatAllOnTimeout                                          | 1      | 0      | OK     | 0.11               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatHandlerOnTimeout                                      | 1      | 0      | OK     | 1.62               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Failure)                                           | 1      | 0      | OK     | 0.2                |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Success)                                           | 1      | 0      | OK     | 0.15               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: CaseNext                                                    | 1      | 0      | OK     | 0.04               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: NoRepeat                                                    | 1      | 0      | OK     | 0.05               |
[... cut ...]
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | tests-netsocket-ip_parsing                                                   | Simple IPv4 address                                                  | 1      | 0      | OK     | 0.05               |
| DISCO_F429ZI-GCC_ARM  | DISCO_F429ZI  | tests-netsocket-ip_parsing                                                   | Simple IPv6 address                                                  | 1      | 0      | OK     | 0.05               |

```
If you have the same results, can we close this issue ?

Kind regards
Armelle

bulislaw, 2016-11-22:
I know that the issue is particularly about GCC, but did you have a chance to check other compilers as well?

adustm, 2016-11-22:
Hello,
Still on Windows 7 host OS:
IAR toolchain 100% OK: 
```
+------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| target           | platform_name | test suite                                                                   | result | elapsed_time (sec) | copy_method |
+------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test                        | OK     | 15.15              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK     | 15.06              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK     | 17.08              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | OK     | 23.11              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK     | 16.83              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_selection                    | OK     | 15.14              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK     | 15.19              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK     | 15.56              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-control_type                      | OK     | 15.88              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK     | 15.59              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK     | 16.42              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK     | 14.48              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK     | 14.79              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK     | 15.32              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | rtos-rtx-target_cortex_m-tests-memory-heap_and_stack                         | OK     | 14.04              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-events-queue                                                           | OK     | 22.16              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-integration-basic                                                      | OK     | 13.85              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-c_strings                                                 | OK     | 17.05              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-dev_null                                                  | OK     | 16.38              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-echo                                                      | OK     | 28.22              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-generic_tests                                             | OK     | 15.2               | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-race_test                                                 | OK     | 15.95              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-rtc                                                       | OK     | 24.75              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-stl_features                                              | OK     | 15.77              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-ticker                                                    | OK     | 48.05              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-timeout                                                   | OK     | 25.47              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_drivers-wait_us                                                   | OK     | 25.69              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_functional-callback                                               | OK     | 16.6               | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_functional-callback_big                                           | OK     | 16.57              | shell       |
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-mbed_functional-callback_small                                         | OK     | 16.51              | shell       |
[... cut ...]
| DISCO_F429ZI-IAR | DISCO_F429ZI  | tests-netsocket-ip_parsing                                                   | Simple IPv6 address                                    | 1| 0      | OK     | 0.05               |
+------------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 164 OK
mbedgt: completed in 903.68 sec
```

adustm, 2016-11-22:
Hello,
Still on Host OS = windows 7
DISCO_F429ZI / ARM toolchain is 100% OK
```
mbedgt: test suite report:
+------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| target           | platform_name | test suite                                                                   | result | elapsed_time (sec) | copy_method |
+------------------+---------------+------------------------------------------------------------------------------+--------+--------------------+-------------+
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test                        | OK     | 15.01              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK     | 15.15              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK     | 18.11              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_async                | OK     | 24.24              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK     | 17.5               | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_selection                    | OK     | 16.23              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK     | 16.82              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK     | 15.41              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-control_type                      | OK     | 16.88              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK     | 15.83              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK     | 18.06              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK     | 15.02              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK     | 14.95              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK     | 16.1               | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | rtos-rtx-target_cortex_m-tests-memory-heap_and_stack                         | OK     | 14.87              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-events-queue                                                           | OK     | 23.7               | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-integration-basic                                                      | OK     | 13.92              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-c_strings                                                 | OK     | 17.86              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-dev_null                                                  | OK     | 16.4               | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-echo                                                      | OK     | 29.72              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-generic_tests                                             | OK     | 15.47              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-race_test                                                 | OK     | 16.46              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-rtc                                                       | OK     | 26.01              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-stl_features                                              | OK     | 17.3               | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-ticker                                                    | OK     | 48.25              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-timeout                                                   | OK     | 25.98              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_drivers-wait_us                                                   | OK     | 25.72              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_functional-callback                                               | OK     | 17.96              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_functional-callback_big                                           | OK     | 16.66              | shell       |
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-mbed_functional-callback_small                                         | OK     | 18.45              | shell       |
[... cut ...]
| DISCO_F429ZI-ARM | DISCO_F429ZI  | tests-netsocket-ip_parsing                                                   | Simple IPv6 address                                    | 1| 0      | OK     | 0.05               |
+------------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 164 OK
mbedgt: completed in 937.19 sec
```

bulislaw, 2016-11-22:
Cool thanks.

closed 2016-11-22 by bulislaw

adustm, 2016-11-22:
You're welcome.
Let me know if you have the same results, so that we can close this issue ... or not...