ARMmbed/mbed-os #2289: ARM_BEETLE_SOC tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: ARM_BEETLE_SOC
Host OS: Windows 7

```

+------------------------+----------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                 | platform_name  | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+------------------------+----------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT | 72.15              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT | 72.15              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 72.12              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT | 72.13              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | TIMEOUT | 72.1               | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | TIMEOUT | 85.97              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | TIMEOUT | 72.1               | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT | 72.08              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | TIMEOUT | 72.13              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 29.58              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 72.11              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-integration-basic                                                      | TIMEOUT | 72.12              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-integration-threaded_blinky                                            | TIMEOUT | 72.16              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-c_strings                                                 | TIMEOUT | 72.11              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-callback                                                  | TIMEOUT | 72.17              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-dev_null                                                  | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-echo                                                      | TIMEOUT | 72.16              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-generic_tests                                             | TIMEOUT | 72.12              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-rtc                                                       | TIMEOUT | 72.09              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-stl_features                                              | TIMEOUT | 72.09              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker                                                    | TIMEOUT | 72.1               | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-timeout                                                   | TIMEOUT | 72.16              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-wait_us                                                   | TIMEOUT | 72.12              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-mbed-attributes                                              | TIMEOUT | 72.13              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | TIMEOUT | 72.16              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-mbed-cpp                                                     | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-mbed-div                                                     | TIMEOUT | 72.12              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | TIMEOUT | 72.09              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | TIMEOUT | 72.07              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | TIMEOUT | 72.16              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | TIMEOUT | 72.14              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 72.13              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | TIMEOUT | 72.15              | shell       |
+------------------------+----------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 OK / 38 TIMEOUT
mbedgt: test case report:
+------------------------+----------------+--------------------------------------------------------------------------------------+----------------------------------------------------------------------
----------------+--------+--------+--------+--------------------+
| target                 | platform_name  | test suite                                                                           | test case
                | passed | failed | result | elapsed_time (sec) |
+------------------------+----------------+--------------------------------------------------------------------------------------+----------------------------------------------------------------------
[... cut ...]
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer
                | 0      | 1      | ERROR  | 72.15              |
+------------------------+----------------+--------------------------------------------------------------------------------------+----------------------------------------------------------------------
----------------+--------+--------+--------+--------------------+
mbedgt: test case results: 8 OK / 38 ERROR
mbedgt: completed in 2784.56 sec

```

0xc0170, 2016-07-28:
cc @fvincenzo

fvincenzo, 2016-07-28:
@MarceloSalazar @abhishek-pandit

To me looks like that the board in your case was not recognized properly.

My configuration:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: ARM_BEETLE_SOC
Host OS: Windows 7

```
mbedgt: test suite report:
+------------------------+----------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                 | platform_name  | test suite                                                                                | result  | elapsed_time (sec) | copy_method |
+------------------------+----------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 20.93              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 21.08              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 22.71              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 29.01              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 22.96              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 20.85              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 21.47              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 21.51              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 21.99              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 21.83              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 22.52              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 20.74              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 20.69              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 20.72              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-integration-basic                                                      | OK      | 19.93              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-integration-threaded_blinky                                            | OK      | 39.91              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-c_strings                                                 | OK      | 22.79              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-callback                                                  | OK      | 27.58              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-dev_null                                                  | OK      | 23.89              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-echo                                                      | OK      | 31.88              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-generic_tests                                             | OK      | 21.78              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-rtc                                                       | OK      | 32.15              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-stl_features                                              | OK      | 22.39              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-ticker                                                    | FAIL    | 80.21              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-timeout                                                   | TIMEOUT | 78.64              | shell       |
[... cut ...]
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-core-tests-mbedmicro-rtos-mbed-timer           | 1      | 0      | OK     | 34.97              |
+------------------------+----------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 131 OK / 2 ERROR
```

fvincenzo, 2016-07-28:
@0xc0170 

I noticed that with the latest rc I have two errors:
- ticker
- timeout

They were working before the latest changes in mbed-os. What did it change?

PrzemekWirkus, 2016-07-28:
@fvincenzo Vincenzo, can you try runnig just these two tests?
```
$ mbedgt -V -n mbed-os-core-tests-mbed_drivers-ticker,mbed-os-core-tests-mbed_drivers-timeout
```
Maybe there is something with the serial, test tools.

fvincenzo, 2016-07-28:
@PrzemekWirkus 

I obtain this:

```
$ mbedgt -V -n mbed-os-core-tests-mbed_drivers-ticker,mbed-os-core-tests-mbed_drivers-timeout
mbedgt: greentea should be run inside a Yotta module or --test-spec switch should be used.
mbedgt: completed in 0.00 sec
mbedgt: exited with code -1
```

PrzemekWirkus, 2016-07-28:
@fvincenzo Maybe you need a newer Greentea, something like v1.0.2 onwards.

This is what I did on mbedmicro:
```
C:\Work\mbed>$ git remote --verbose
origin  https://github.com/mbedmicro/mbed.git (fetch)
origin  https://github.com/mbedmicro/mbed.git (push)
```
```
C:\Work\mbed>$ mbed test --compile -t GCC_ARM -m K64F
```
```
C:\Work\mbed>$mbedgt -V -n tests-mbed_drivers-timeout
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: using multiple test specifications from current directory!
        using '.build\tests\K64F\GCC_ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
```

PrzemekWirkus, 2016-07-28:
Oh, I think the test names changed...

fvincenzo, 2016-07-28:
This is what I see (it works now :) ):
```
$ mbedgt -V -n mbed-os-core-tests-mbed_drivers-ticker
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: using multiple test specifications from current directory!
        using '.build\tests\ARM_BEETLE_SOC\GCC_ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +----------------+----------------------+-------------+-------------+--------------------------+
        | platform_name  | platform_name_unique | serial_port | mount_point | target_id                |
        +----------------+----------------------+-------------+-------------+--------------------------+
        | ARM_BEETLE_SOC | ARM_BEETLE_SOC[0]    | COM27       | F:          | 500302030C48FD3AF1B452C1 |
        +----------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'ARM_BEETLE_SOC' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +----------------+----------------------+-------------+-------------+--------------------------+
        | platform_name  | platform_name_unique | serial_port | mount_point | target_id                |
        +----------------+----------------------+-------------+-------------+--------------------------+
        | ARM_BEETLE_SOC | ARM_BEETLE_SOC[0]    | COM27:9600  | F:          | 500302030C48FD3AF1B452C1 |
        +----------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'mbed-os-core-tests-mbed_drivers-ticker'
mbedgt: running 1 test for platform 'ARM_BEETLE_SOC' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m ARM_BEETLE_SOC -p COM27:9600 -f ".build/tests/ARM_BEETLE_SOC/GCC_ARM/mbed-os/core/TESTS/mbed_drivers/ticker/ticker.bin" -d F: -C 4 -c shell -t 500302030C48FD3AF1B452C1
mbedgt: mbed-host-test-runner: started
[1469714254.65][HTST][INF] host test executor ver. 1.1.0
[1469714254.65][HTST][INF] copy image onto target...
[1469714254.65][COPY][INF] Waiting up to 60 sec for '500302030C48FD3AF1B452C1' mount point (current is 'F:')...
        1 file(s) copied.
[1469714272.81][HTST][INF] starting host test process...
[1469714273.14][CONN][INF] starting connection process...
[1469714273.14][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1469714273.14][CONN][INF] initializing serial port listener...
[1469714273.15][HTST][INF] setting timeout to: 60 sec
[1469714273.15][SERI][INF] serial(port=COM27, baudrate=9600, timeout=0)
[1469714273.15][PLGN][INF] Waiting up to 60 sec for '500302030C48FD3AF1B452C1' serial port (current is 'COM27')...
[1469714273.19][SERI][INF] reset device using 'default' plugin...
[1469714273.44][SERI][INF] waiting 1.00 sec after reset
[1469714274.44][SERI][INF] wait for it...
[1469714274.44][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1469714274.44][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1469714274.44][CONN][INF] sending preamble '846a47d7-3553-40e5-affe-82c1a9e899c6'
[1469714274.44][SERI][TXD] {{__sync;846a47d7-3553-40e5-affe-82c1a9e899c6}}
[1469714274.58][CONN][RXD] {{__sync;846a47d7-3553-40e5-affe-82c1a9e899c6}}
[1469714274.58][CONN][INF] found SYNC in stream: {{__sync;846a47d7-3553-40e5-affe-82c1a9e899c6}} it is #0 sent, queued...
[1469714274.58][HTST][INF] sync KV found, uuid=846a47d7-3553-40e5-affe-82c1a9e899c6, timestamp=1469714274.584000
[1469714274.60][CONN][RXD] {{__version;1.3.0}}
[1469714274.61][HTST][INF] DUT greentea-client version: 1.3.0
[1469714274.61][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1469714274.62][CONN][RXD] {{__timeout;60}}
[1469714274.62][HTST][INF] setting timeout to: 60 sec
[1469714274.62][CONN][INF] found KV pair in stream: {{__timeout;60}}, queued...
[1469714274.66][CONN][RXD] {{__host_test_name;wait_us_auto}}
[1469714274.66][CONN][INF] found KV pair in stream: {{__host_test_name;wait_us_auto}}, queued...
[1469714274.66][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.wait_us_auto.WaitusTest'>'
[1469714274.66][HTST][INF] host test setup() call...
[1469714274.66][HTST][INF] CALLBACKs updated
[1469714274.66][HTST][INF] host test detected: wait_us_auto
[1469714274.68][CONN][RXD] {{__testcase_count;3}}
[1469714274.68][CONN][INF] found KV pair in stream: {{__testcase_count;3}}, queued...
[1469714274.71][CONN][RXD] >>> Running 3 test cases...
[1469714274.77][CONN][RXD] {{__testcase_name;Timers: 1x ticker}}
[1469714274.77][CONN][INF] found KV pair in stream: {{__testcase_name;Timers: 1x ticker}}, queued...
[1469714274.79][CONN][RXD] {{__testcase_name;Timers: 2x tickers}}
[1469714274.79][CONN][INF] found KV pair in stream: {{__testcase_name;Timers: 2x tickers}}, queued...
[1469714274.83][CONN][RXD] {{__testcase_name;Timers: 2x callbacks}}
[1469714274.83][CONN][INF] found KV pair in stream: {{__testcase_name;Timers: 2x callbacks}}, queued...
[... 67 lines cut ...]
mbedgt: test suite 'mbed-os-core-tests-mbed_drivers-ticker' .......................................... FAIL in 80.31 sec
[... cut ...]
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-ticker | Timers: 2x callbacks | 0      | 0      | SKIPPED | 0.0                |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-core-tests-mbed_drivers-ticker | Timers: 2x tickers   | 0      | 0      | ERROR   | 0.0                |
+------------------------+----------------+----------------------------------------+----------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 1 SKIPPED / 1 OK / 1 ERROR
mbedgt: completed in 80.38 sec
mbedgt: exited with code 1
```

PrzemekWirkus, 2016-07-28:
@fvincenzo I think someone refactored timer test to contain few timer/ticker tests. You fail on second one (Timers: 2x tickers).

fvincenzo, 2016-07-28:
@PrzemekWirkus I will re-test when the test strategy on timers will be reported to the old one. Thanks for you help :+1:

bridadan, 2016-07-28:
@fvincenzo Thanks for reporting this. Like @PrzemekWirkus said, someone (me actually!) consolidated some the ticker tests to speed up testing a bit and to reduce the amount of duplicated code between the tests. This is the consolidated test case now: https://github.com/mbedmicro/mbed/blob/master/TESTS/mbed_drivers/ticker/main.cpp

The interesting thing from your test output is that the second test case inside the consolidate test stops halfway through, which seems like a legitimate issue. If you want, we can try and look at this offline on Skype or something.

fvincenzo, 2016-07-28:
@bridadan it is ok for me, we can look at it if you think that it is a legitimate issue.

fvincenzo, 2016-08-12:
@MarceloSalazar @bridadan @screamerbg I verified the tests with the latest mbed-os master code changes (PR: https://github.com/ARMmbed/mbed-os/pull/2402) and everything works correctly. I think this ticket can be closed:
```
mbedgt: test suite report:
+------------------------+----------------+------------------------------------+--------+--------------------+-------------+
| target                 | platform_name  | test suite                         | result | elapsed_time (sec) | copy_method |
+------------------------+----------------+------------------------------------+--------+--------------------+-------------+
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker  | OK     | 52.07              | shell       |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-timeout | OK     | 32.27              | shell       |
+------------------------+----------------+------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 2 OK
mbedgt: test case report:
+------------------------+----------------+------------------------------------+-----------------------+--------+--------+--------+--------------------+
| target                 | platform_name  | test suite                         | test case             | passed | failed | result | elapsed_time (sec) |
+------------------------+----------------+------------------------------------+-----------------------+--------+--------+--------+--------------------+
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker  | Timers: 1x ticker     | 1      | 0      | OK     | 11.04              |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker  | Timers: 2x callbacks  | 1      | 0      | OK     | 10.04              |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-ticker  | Timers: 2x tickers    | 1      | 0      | OK     | 10.04              |
| ARM_BEETLE_SOC-GCC_ARM | ARM_BEETLE_SOC | mbed-os-tests-mbed_drivers-timeout | Timers: toggle on/off | 1      | 0      | OK     | 11.83              |
+------------------------+----------------+------------------------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 4 OK
mbedgt: completed in 84.41 sec
'''

MarceloSalazar, 2016-08-12:
tested master (https://github.com/ARMmbed/mbed-os/#685096eb11af42db747273415aa1d5e2ad94f2ab) and tests pass when building with gcc.
closing

closed 2016-08-12 by MarceloSalazar