ARMmbed/mbed-os #2290: Migration documentation minar -> threads

markus-becker-tridonic-com, 2016-07-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type:  Question
- Priority:  Minor

## Question

Where do I find the documentation on how to move from minar events to RTOS threads?

Re-opened from https://github.com/ARMmbed/mbed-os/issues/445#issuecomment-235424621.

@MarceloSalazar

sg-, 2016-08-26:
There will be docs and a path for this part of the mbed OS 5.2 release

MarceloSalazar, 2016-10-19:
This document should clarify the questions:
https://github.com/ARMmbed/Handbook/blob/5.2/docs/advanced/MINAR_migration.md

If something is not clear / missing, please raise an issue on the doc's repo.

closed 2016-10-19 by MarceloSalazar