ARMmbed/mbed-os #2291: NUCLEO_L476RG tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: NUCLEO_L476RG
Host OS: Linux

```

+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.95               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT | 64.61              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 64.32              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT | 64.63              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.93              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.87               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.42               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | TIMEOUT | 64.62              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT | 67.15              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.74               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 12.41              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 64.61              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-integration-basic                                                      | TIMEOUT | 64.59              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-integration-threaded_blinky                                            | TIMEOUT | 64.77              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-c_strings                                                 | TIMEOUT | 64.62              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-callback                                                  | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-dev_null                                                  | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-echo                                                      | TIMEOUT | 64.61              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-generic_tests                                             | TIMEOUT | 67.14              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-rtc                                                       | TIMEOUT | 67.18              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-stl_features                                              | TIMEOUT | 67.14              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-ticker                                                    | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-timeout                                                   | TIMEOUT | 64.61              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbed_drivers-wait_us                                                   | TIMEOUT | 64.61              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-mbed-attributes                                              | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.38               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-mbed-cpp                                                     | TIMEOUT | 64.63              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-mbed-div                                                     | OK      | 8.51               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | TIMEOUT | 64.62              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 13.5               | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.66              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.82              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | TIMEOUT | 64.33              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 64.63              | shell       |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 23.09              | shell       |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 12 OK / 27 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                                                                           | test case                                                                            | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
[... cut ...]
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | 1                        | 23.09              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+------                ----+--------------------+
mbedgt: test case results: 30 OK / 27 ERROR
mbedgt: completed in 1900.47 sec

```

screamerbg, 2016-07-28:
CC @adustm

bridadan, 2016-07-28:
I mentioned this in #2286, but I'll go ahead and copy it below:

> There is an issue with ST-Link on Linux where the disk drive appears to "fill up" after multiple binaries have been flashed to the board. This causes flashing to fail with the testing tools and requires the board to be manually unplugged then plugged back into the machine. @adustm we have raised this issue before, have you heard anything about this issue recently?

adustm, 2016-11-22:
Hello @bridadan,
The test results on mbed-os-5.2.2 are good for NUCLEO_L476RG with GCC_ARM). Only LP timer tests are failed (already mentionned in another issue).
The Host-OS is Windows 7.
Could you try the update of ST-Link ? It is supposed to fix the issue for linux.
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link007.html
Let us know
Cheers
Armelle

adustm, 2017-05-05:
Hello,
Here is the test report for NUCLEO-L476RG on mbed-os-5.4.3
Is it the same on your side ? Do you agree to close this issue ?
Cheers
Armelle
```
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                         | 2      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                            | 1      | 0      | OK     | 0.03               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-basic_test_default                | Repeating Test                                         | 2      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-basic_test_default                | Simple Test                                            | 1      | 0      | OK     | 0.03               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Cancel Repeat         | 1      | 0      | OK     | 0.18               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Enable Repeat Handler | 2      | 0      | OK     | 0.19               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Premature Validation                | 1      | 0      | OK     | 0.08               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Validation                          | 1      | 0      | OK     | 0.16               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Premature Validation                         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Simple Validation                            | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Await                                         | 1      | 0      | OK     | 1.42               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: CaseNext                                      | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: NoTimeout                                     | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatAllOnTimeout                            | 1      | 0      | OK     | 0.11               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatHandlerOnTimeout                        | 1      | 0      | OK     | 1.62               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Failure)                             | 1      | 0      | OK     | 0.2                |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Success)                             | 1      | 0      | OK     | 0.16               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: CaseNext                                      | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: NoRepeat                                      | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: RepeatAll                                     | 10     | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: RepeatHandler                                 | 10     | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_selection                    | Case 1                                                 | 1      | 0      | OK     | 0.03               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | features-frameworks-utest-tests-unit_tests-case_selection                    | Case 2                                                 | 1      | 0      | OK     | 0.05               |
[... cut ...]
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-netsocket-ip_parsing                                                   | Simple IPv4 address                                    | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-netsocket-ip_parsing                                                   | Simple IPv6 address                                    | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-st_internal-st_hw_entropy                                              | tests-st_internal-st_hw_entropy                        | 1      | 0      | OK     | 66.11              |
```

bridadan, 2017-05-10:
Just ran all tests on master as well, all passing on all three toolchains. Closing this, thanks @adustm .

closed 2017-05-10 by bridadan