ARMmbed/mbed-os #2292: NUCLEO_F411RE tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: NUCLEO_F411RE
Host OS: Windows 7

```

+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT | 72.57              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.26               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | TIMEOUT | 72.43              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 17.14              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 11.15              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.11               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.71               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.84               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.91               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 10.03              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.65              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT | 72.47              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 185.6              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 9.03               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-integration-basic                                                      | OK      | 8.42               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-integration-threaded_blinky                                            | TIMEOUT | 72.63              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 11.05              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-callback                                                  | OK      | 15.35              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-dev_null                                                  | TIMEOUT | 72.6               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 21.38              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-generic_tests                                             | OK      | 9.57               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-rtc                                                       | OK      | 29.18              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-stl_features                                              | OK      | 10.24              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-ticker                                                    | TIMEOUT | 72.69              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 20.08              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.94              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-attributes                                              | TIMEOUT | 72.43              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.57               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.71               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-div                                                     | OK      | 8.67               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | TIMEOUT | 72.47              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 13.74              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | TIMEOUT | 72.5               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 19.15              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT | 72.45              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 16.28              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | TIMEOUT | 72.38              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 77.49              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 23.39              | shell       |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 26 OK / 13 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                                                                           | test case                                                                            | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
[... cut ...]
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | 1      | 0      | OK     | 23.39              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 89 OK / 14 ERROR
mbedgt: completed in 1410.38 sec
```

screamerbg, 2016-07-28:
CC @adustm

MarceloSalazar, 2016-07-30:
Tested RC4
All tests pass

```
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK     | 8.84               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK     | 8.86               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK     | 10.49              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK     | 16.71              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK     | 10.81              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK     | 8.7                | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK     | 9.27               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK     | 9.36               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK     | 9.44               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK     | 9.68               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK     | 10.35              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK     | 8.49               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK     | 8.67               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK     | 8.67               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-integration-basic                                                      | OK     | 8.04               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-integration-threaded_blinky                                            | OK     | 28.01              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-c_strings                                                 | OK     | 10.54              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-callback                                                  | OK     | 14.91              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-dev_null                                                  | OK     | 10.92              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-echo                                                      | OK     | 20.27              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-generic_tests                                             | OK     | 9.23               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-rtc                                                       | OK     | 20.23              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-stl_features                                              | OK     | 9.87               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-ticker                                                    | OK     | 39.89              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-timeout                                                   | OK     | 19.7               | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbed_drivers-wait_us                                                   | OK     | 18.53              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK     | 10.57              | shell       |
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK     | 8.19               | shell       |
[... cut ...]
| NUCLEO_F411RE-GCC_ARM | NUCLEO_F411RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                | 1      | 0      | OK     | 22.97
            |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------
------------+
mbedgt: test case results: 134 OK
mbedgt: completed in 525.40 sec
```

closed 2016-07-30 by MarceloSalazar