ARMmbed/mbed-os #2293: NUCLEO_F746ZG tests fail - GCC OoB RC#1

MarceloSalazar, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: NUCLEO_F746ZG
Host OS: Windows 7

```

+-----------------------+---------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK     | 8.47               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK     | 8.44               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK     | 10.13              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK     | 16.31              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK     | 10.41              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK     | 8.33               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK     | 8.89               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK     | 8.99               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK     | 9.25               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK     | 9.23               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK     | 9.97               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK     | 8.1                | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK     | 8.27               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK     | 8.29               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-integration-basic                                                      | OK     | 7.67               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-integration-threaded_blinky                                            | OK     | 27.63              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-c_strings                                                 | OK     | 10.2               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-callback                                                  | OK     | 14.22              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-dev_null                                                  | OK     | 10.21              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-echo                                                      | OK     | 20.05              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-generic_tests                                             | OK     | 9.04               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-rtc                                                       | OK     | 28.12              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-stl_features                                              | OK     | 9.13               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-ticker                                                    | OK     | 39.49              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-timeout                                                   | OK     | 19.2               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbed_drivers-wait_us                                                   | OK     | 18.14              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK     | 10.16              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK     | 7.79               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-mbed-cpp                                                     | OK     | 7.91               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-mbed-div                                                     | OK     | 7.93               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-basic                                              | OK     | 22.62              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-isr                                                | OK     | 12.87              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-mail                                               | OK     | 9.32               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-mutex                                              | OK     | 18.01              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-queue                                              | OK     | 9.34               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-semaphore                                          | OK     | 15.17              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-signals                                            | OK     | 12.92              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-threads                                            | OK     | 13.47              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | OK     | 22.63              | shell       |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 39 OK
mbedgt: test case report:
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+--------+--------------------+
| target                | platform_name | test suite                                                                           | test case                                              | passed | faile
d | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+--------+--------------------+
[... cut ...]
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                | 1      | 0
  | OK     | 22.63              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+--------+--------------------+
mbedgt: test case results: 133 OK / 1 ERROR
mbedgt: completed in 516.90 sec

```

screamerbg, 2016-07-28:
cc @adustm

MarceloSalazar, 2016-07-30:
RC#4

```
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                           | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 8.84               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.9                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 10.58              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.71              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.84              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.72               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.27               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.33               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.44               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.59               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.31              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.5                | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.65               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.69               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-basic                                                      | OK      | 8.03               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-integration-threaded_blinky                                            | OK      | 28.02              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-c_strings                                                 | OK      | 10.53              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-callback                                                  | OK      | 14.89              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-dev_null                                                  | OK      | 10.91              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-echo                                                      | OK      | 20.49              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-generic_tests                                             | OK      | 9.25               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-rtc                                                       | OK      | 20.03              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-stl_features                                              | OK      | 9.87               | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-ticker                                                    | OK      | 39.95              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-timeout                                                   | OK      | 19.65              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbed_drivers-wait_us                                                   | OK      | 18.53              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-attributes                                              | OK      | 10.54              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.19               | shell       |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | mbed-os-tests-mbedmicro-rtos-mbed-timer                                              | mbed-os-tests-mbedmicro-rtos-mbed-timer                | 1      | 0
  | OK      | 23.01              |
+-----------------------+---------------+--------------------------------------------------------------------------------------+--------------------------------------------------------+--------+------
--+---------+--------------------+
mbedgt: test case results: 13 SKIPPED / 120 OK / 1 ERROR
mbedgt: completed in 558.59 sec

```

bulislaw, 2016-08-02:
OS: OSX
Toolchain: GCC
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc4
```
mbedgt: test suite report:
+-----------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT | 66.09              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 8.91               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 10.54              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 16.72              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 10.78              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 8.65               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.29               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.53               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT | 66.2               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.88               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | TIMEOUT | 66.16              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 8.48               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 8.64               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 8.69               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-integration-basic                                                      | TIMEOUT | 64.96              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-integration-threaded_blinky                                            | OK      | 27.96              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-c_strings                                                 | OK      | 10.55              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-callback                                                  | TIMEOUT | 66.85              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-dev_null                                                  | TIMEOUT | 66.64              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-echo                                                      | OK      | 20.23              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-generic_tests                                             | TIMEOUT | 66.23              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-rtc                                                       | TIMEOUT | 64.87              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-stl_features                                              | OK      | 9.63               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-ticker                                                    | OK      | 39.84              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-timeout                                                   | OK      | 19.54              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbed_drivers-wait_us                                                   | OK      | 18.52              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbedmicro-mbed-attributes                                              | OK      | 10.64              | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.25               | shell       |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-mbedmicro-mbed-cpp                                                     | TIMEOUT | 66.16              | shell       |
[... cut ...]
+-----------------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 88 OK / 13 ERROR
mbedgt: completed in 1231.62 sec
mbedgt: exited with code 13
[mbed] ERROR: "mbedgt" returned error code 13.
[mbed] ERROR: Command "mbedgt --test-spec /Users/barsza01/devel/mbed/mbed-os/.build/tests/NUCLEO_F746ZG/GCC_ARM/test_spec.json" in "/Users/barsza01/devel/mbed/mbed-os"
---
```

adustm, 2017-05-05:
Hello,
Here is the test report of the NUCLEO_F746ZG GCC_ARM on the mbed-os-5.4.3
Do you have the same status on your side ? Would you agree to close this issue ?
```
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-connectivity                       | Testing bringing the network up and down                     | 1      | 0      | OK     | 3.71               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-connectivity                       | Testing bringing the network up and down twice               | 1      | 0      | OK     | 1.81               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-gethostbyname                      | Testing DNS literal                                          | 1      | 0      | OK     | 0.1                |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-gethostbyname                      | Testing DNS preference literal                               | 1      | 0      | OK     | 0.13               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-gethostbyname                      | Testing DNS preference query                                 | 1      | 0      | OK     | 0.1                |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-gethostbyname                      | Testing DNS query                                            | 1      | 0      | OK     | 0.07               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-tcp_echo                           | features-feature_lwip-tests-mbedmicro-net-tcp_echo           | 1      | 0      | OK     | 78.2               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-tcp_echo_parallel                  | features-feature_lwip-tests-mbedmicro-net-tcp_echo_parallel  | 1      | 0      | OK     | 80.93              |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-tcp_hello_world                    | features-feature_lwip-tests-mbedmicro-net-tcp_hello_world    | 1      | 0      | OK     | 78.13              |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake                 | features-feature_lwip-tests-mbedmicro-net-udp_dtls_handshake | 1      | 0      | OK     | 77.73              |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-udp_echo                           | features-feature_lwip-tests-mbedmicro-net-udp_echo           | 1      | 0      | OK     | 78.95              |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-udp_echo_parallel                  | features-feature_lwip-tests-mbedmicro-net-udp_echo_parallel  | 1      | 0      | OK     | 79.65              |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                               | 2      | 0      | OK     | 0.08               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                                  | 1      | 0      | OK     | 0.04               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test_default                | Repeating Test                                               | 2      | 0      | OK     | 0.08               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-basic_test_default                | Simple Test                                                  | 1      | 0      | OK     | 0.03               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Cancel Repeat               | 1      | 0      | OK     | 0.17               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Enable Repeat Handler       | 2      | 0      | OK     | 0.18               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Premature Validation                      | 1      | 0      | OK     | 0.08               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Validation                                | 1      | 0      | OK     | 0.15               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Premature Validation                               | 1      | 0      | OK     | 0.06               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Simple Validation                                  | 1      | 0      | OK     | 0.07               |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Await                                               | 1      | 0      | OK     | 1.42                [... cut ...]                |
| NUCLEO_F746ZG-GCC_ARM | NUCLEO_F746ZG | tests-st_internal-st_hw_entropy                                              | tests-st_internal-st_hw_entropy                              | 1      | 0      | OK     | 74.21              |
| NUCLEO_F746ZG-IAR     | NUCLEO_F746ZG | features-feature_lwip-tests-mbedmicro-net-connectivity                       | Testing bringing the network up and down                     | 1      | 0      | OK     | 3.7                |
```

Cheers
Armelle

jeromecoutant, 2017-05-22:
ST_TO_BE_CLOSED

MarceloSalazar, 2017-05-22:
I've tested mbed OS 5.4.5 (8d21974ba35e04c4854e5090c0f8283171664175). 
All tests are passing for GCC_ARM and ARM Compiler.

For IAR, there are two tests failing:

 ```
+-------------------+---------------+----------------------------------------------------------------+--------+--------------------+-------------+
| target            | platform_name | test suite                                                     | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+----------------------------------------------------------------+--------+--------------------+-------------+
| NUCLEO_F746ZG-IAR | NUCLEO_F746ZG | mbed-os-features-feature_lwip-tests-mbedmicro-net-connectivity | FAIL   | 26.05              | shell       |
+-------------------+---------------+----------------------------------------------------------------+--------+--------------------+-------------+

+-------------------+---------------+------------------------------------------------------------+--------+--------------------+-------------+
| target            | platform_name | test suite                                                 | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+------------------------------------------------------------+--------+--------------------+-------------+
| NUCLEO_F746ZG-IAR | NUCLEO_F746ZG | mbed-os-features-feature_lwip-tests-mbedmicro-net-udp_echo | FAIL   | 15.87              | shell       |
+-------------------+---------------+------------------------------------------------------------+--------+--------------------+-------------+
```
@adustm @jeromecoutant could you please have a look and let me know if you are unable to reproduce?

bcostm, 2017-10-16:
I have checked in our test result database and both tests are PASS from mbed-os versions 5.5.1 to 5.5.6

closed 2017-10-16 by MarceloSalazar