ARMmbed/mbed-os #2296: EFM32PG_STK3401 tests fail - GCC OoB RC#1

bulislaw, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
GCC toolchain
Target: EFM32PG_STK3401 (Pearl Gecko)
Host OS: Linux

```
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| target                  | platform_name   | test suite                                                                   | result       | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | OK           | 8.49               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | IOERR_SERIAL | 8.4                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | IOERR_SERIAL | 7.47               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | IOERR_SERIAL | 10.28              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | IOERR_SERIAL | 8.69               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | OK           | 8.53               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK           | 8.6                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK           | 8.63               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | OK           | 8.82               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK           | 8.72               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK           | 8.67               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK           | 8.4                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK           | 8.39               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT      | 66.23              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | OK           | 8.35               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | IOERR_SERIAL | 9.44               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | OK           | 8.69               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | OK           | 9.51               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | TIMEOUT      | 66.95              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | TIMEOUT      | 66.24              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | OK           | 8.79               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | IOERR_SERIAL | 9.83               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | TIMEOUT      | 66.68              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | IOERR_SERIAL | 10.45              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | OK           | 28.45              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | IOERR_SERIAL | 13.48              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | TIMEOUT      | 64.81              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-call_before_main                                        | OK           | 8.5                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-cpp                                                     | TIMEOUT      | 66.4               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-div                                                     | TIMEOUT      | 66.37              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-basic                                              | IOERR_SERIAL | 13.33              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-isr                                                | IOERR_SERIAL | 8.58               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-mail                                               | OK           | 10.07              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-mutex                                              | IOERR_SERIAL | 8.66               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-queue                                              | IOERR_SERIAL | 9.06               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-semaphore                                          | IOERR_SERIAL | 9.88               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-signals                                            | OK           | 13.63              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-threads                                            | OK           | 10.13              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-timer                                              | FAIL         | 23.47              | shell       |
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
+-------------------------+-----------------+------------------------------------------------------------------------------+----------------------------------------------------------------+--------+--------+---------+--------------------+
| target                  | platform_name   | test suite                                                                   | test case                                                      | passed | failed | result  | elapsed_time (sec) |
+-------------------------+-----------------+------------------------------------------------------------------------------+----------------------------------------------------------------+--------+--------+---------+--------------------+
[... cut ...]
+-------------------------+-----------------+------------------------------------------------------------------------------+----------------------------------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 1 FAIL / 12 SKIPPED / 83 OK / 21 ERROR
mbedgt: completed in 798.45 sec
mbedgt: exited with code 21
[mbed] ERROR: "mbedgt" returned error code 21.
[mbed] ERROR: Command "mbedgt --test-spec /work/mbed/mbed-os/.build/tests/EFM32PG_STK3401/GCC_ARM/test_spec.json" in "/work/mbed/mbed-os"
```

bulislaw, 2016-07-29:
For Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc3
```
mbedgt: test suite report:
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| target                  | platform_name   | test suite                                                                   | result       | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT      | 69.75              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK           | 7.58               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | IOERR_SERIAL | 8.7                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | IOERR_SERIAL | 10.46              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK           | 8.62               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | IOERR_SERIAL | 8.5                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK           | 8.58               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK           | 8.55               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | TIMEOUT      | 23.69              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK           | 8.77               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | IOERR_SERIAL | 7.45               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK           | 8.45               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | IOERR_SERIAL | 8.44               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK           | 8.49               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | OK           | 8.27               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | FAIL         | 28.29              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | TIMEOUT      | 13.43              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | OK           | 9.56               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | IOERR_SERIAL | 9.01               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | FAIL         | 188.47             | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | OK           | 8.79               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | OK           | 18.92              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | IOERR_SERIAL | 8.9                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | FAIL         | 68.37              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | IOERR_SERIAL | 8.39               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | IOERR_SERIAL | 9.43               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | IOERR_SERIAL | 8.5                | shell       |
[... cut ...]
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-timer                                              | tests-mbedmicro-rtos-mbed-timer                                              | 0      | 1      | ERROR   | 8.26   
            |
+-------------------------+-----------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+---------+--------
------------+
```

bridadan, 2016-08-02:
@bulislaw I unfortunately don't have one of these targets to test with. Is there any chance you have the verbose output log of this? It can be ran with `mbed test -v`.

bulislaw, 2016-08-03:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc5
```
mbedgt: test suite report:
+-------------------------+-----------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target                  | platform_name   | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+-------------------------+-----------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 9.41               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.3                | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 9.85               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 13.37              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 9.55               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.29               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.39               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.48               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 9.58               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 9.85               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 9.46               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 9.25               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 9.24               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 9.44               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | OK      | 9.18               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | OK      | 29.0               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | OK      | 9.67               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | OK      | 10.34              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | OK      | 11.92              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | OK      | 11.28              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | OK      | 9.61               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | OK      | 19.84              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | OK      | 10.04              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | FAIL    | 69.27              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | OK      | 28.66              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | OK      | 19.62              | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | OK      | 9.56               | shell       |
| EFM32PG_STK3401-GCC_ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-call_before_main                                        | OK      | 9.36               | shell       |
[... cut ...]
+-------------------------+-----------------+------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 131 OK / 3 ERROR
mbedgt: completed in 612.60 sec
mbedgt: exited with code 2
[mbed] ERROR: "mbedgt" returned error code 2.
[mbed] ERROR: Command "mbedgt --test-spec /Users/barsza01/devel/mbed/mbed-os/.build/tests/EFM32PG_STK3401/GCC_ARM/test_spec.json" in "/Users/barsza01/devel/mbed/mbed-os"
---
```

bulislaw, 2016-08-03:
@bridadan sure will do that later.

closed 2017-10-27 by ChiefBureaucraticOfficer