ARMmbed/mbed-os #2299: EFM32PG_STK3401 tests fail - ARM OoB RC#1

bulislaw, 2016-07-28:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc1
ARM toolchain
Target: EFM32PG_STK3401 (Pearl Gecko)
Host OS: Linux
```
mbedgt: test suite report:
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| target              | platform_name   | test suite                                                                   | result       | elapsed_time (sec) | copy_method |
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | OK           | 7.98               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT      | 17.92              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK           | 8.43               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT      | 65.73              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | IOERR_SERIAL | 8.09               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | IOERR_SERIAL | 8.03               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | IOERR_SERIAL | 7.93               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | IOERR_SERIAL | 8.04               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | OK           | 8.26               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK           | 8.16               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | IOERR_SERIAL | 8.1                | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | IOERR_SERIAL | 6.92               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK           | 7.9                | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | IOERR_SERIAL | 7.94               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | OK           | 7.84               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | FAIL         | 27.75              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | TIMEOUT      | 65.77              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | IOERR_SERIAL | 8.56               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | OK           | 9.79               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | FAIL         | 187.85             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | TIMEOUT      | 65.78              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | OK           | 18.07              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | TIMEOUT      | 66.23              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | IOERR_SERIAL | 9.88               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | IOERR_SERIAL | 7.86               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | IOERR_SERIAL | 10.9               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | IOERR_SERIAL | 7.95               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-call_before_main                                        | OK           | 7.89               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-cpp                                                     | TIMEOUT      | 65.53              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-div                                                     | IOERR_SERIAL | 7.86               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-basic                                              | FAIL         | 22.77              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-isr                                                | OK           | 12.8               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-mail                                               | IOERR_SERIAL | 8.06               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-mutex                                              | IOERR_SERIAL | 9.76               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-queue                                              | TIMEOUT      | 64.67              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-semaphore                                          | IOERR_SERIAL | 10.93              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-signals                                            | TIMEOUT      | 65.71              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT      | 64.81              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-rtos-mbed-timer                                              | IOERR_SERIAL | 15.84              | shell       |
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
mbedgt: test suite results: 3 FAIL / 10 OK / 17 IOERR_SERIAL / 9 TIMEOUT
mbedgt: test case report:
+---------------------+-----------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+---------+------------
--------+
| target              | platform_name   | test suite                                                                   | test case                                                                    | passed | failed | result  | elapsed_tim
e (sec) |
+---------------------+-----------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+---------+------------
--------+
[... cut ...]
+---------------------+-----------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+---------+------------
--------+
mbedgt: test case results: 2 FAIL / 28 SKIPPED / 41 OK / 18 ERROR
mbedgt: completed in 1030.58 sec
mbedgt: exited with code 29
[mbed] ERROR: "mbedgt" returned error code 29.
[mbed] ERROR: Command "mbedgt --test-spec /work/mbed/mbed-os/.build/tests/EFM32PG_STK3401/ARM/test_spec.json" in "/work/mbed/mbed-os"
---
```

bulislaw, 2016-07-29:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc3
```
mbedgt: test suite report:
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| target              | platform_name   | test suite                                                                   | result       | elapsed_time (sec) | copy_method |
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT      | 137.14             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT      | 137.24             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | IOERR_SERIAL | 8.36               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT      | 137.26             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | TIMEOUT      | 136.67             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | OK           | 7.98               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK           | 7.93               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | TIMEOUT      | 137.28             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | IOERR_SERIAL | 8.11               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | TIMEOUT      | 137.74             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | TIMEOUT      | 136.51             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT      | 136.65             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT      | 136.43             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK           | 7.95               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | TIMEOUT      | 137.24             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | TIMEOUT      | 136.54             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | IOERR_SERIAL | 8.12               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | TIMEOUT      | 137.42             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | TIMEOUT      | 136.68             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | TIMEOUT      | 136.96             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | TIMEOUT      | 136.56             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | TIMEOUT      | 137.42             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | TIMEOUT      | 137.43             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | TIMEOUT      | 136.62             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | TIMEOUT      | 136.63             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | TIMEOUT      | 137.46             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | TIMEOUT      | 137.47             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-call_before_main                                        | OK           | 7.78               | shell       |
[... cut ...]
+---------------------+-----------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+---------+------------
--------+
mbedgt: test case results: 6 SKIPPED / 14 OK / 36 ERROR
mbedgt: completed in 4017.73 sec
mbedgt: exited with code 33
[mbed] ERROR: "mbedgt" returned error code 33.
[mbed] ERROR: Command "mbedgt --test-spec /work/mbed/mbed-os/.build/tests/EFM32PG_STK3401/ARM/test_spec.json" in "/work/mbed/mbed-os"
---
```

bulislaw, 2016-08-03:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc5
```
mbedgt: test suite report:
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| target              | platform_name   | test suite                                                                   | result       | elapsed_time (sec) | copy_method |
+---------------------+-----------------+------------------------------------------------------------------------------+--------------+--------------------+-------------+
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-basic_test_default                | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK           | 8.4                | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_control_repeat               | TIMEOUT      | 137.65             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_selection                    | OK           | 7.83               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK           | 7.96               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-control_type                      | OK           | 8.12               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | IOERR_SERIAL | 7.91               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK           | 7.91               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | IOERR_SERIAL | 7.82               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK           | 7.91               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-basic                                                      | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-integration-threaded_blinky                                            | OK           | 27.73              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-c_strings                                                 | IOERR_SERIAL | 8.13               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-callback                                                  | TIMEOUT      | 137.65             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-dev_null                                                  | OK           | 9.82               | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-echo                                                      | IOERR_SERIAL | 7.9                | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-generic_tests                                             | TIMEOUT      | 69.11              | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-rtc                                                       | TIMEOUT      | 137.65             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-stl_features                                              | TIMEOUT      | 137.87             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-ticker                                                    | TIMEOUT      | 137.65             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-timeout                                                   | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbed_drivers-wait_us                                                   | TIMEOUT      | 137.76             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-attributes                                              | TIMEOUT      | 138.09             | shell       |
| EFM32PG_STK3401-ARM | EFM32PG_STK3401 | tests-mbedmicro-mbed-call_before_main                                        | TIMEOUT      | 64.6               | shell       |
[... cut ...]
+---------------------+-----------------+------------------------------------------------------------------------------+--------------------------------------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 8 SKIPPED / 31 OK / 26 ERROR
mbedgt: completed in 2987.15 sec
mbedgt: exited with code 28
[mbed] ERROR: "mbedgt" returned error code 28.
[mbed] ERROR: Command "mbedgt --test-spec /work/mbed/mbed-os/.build/tests/EFM32PG_STK3401/ARM/test_spec.json" in "/work/mbed/mbed-os"
---

```

sg-, 2017-01-16:
These issues look a lot like interface firmware failures. Is the firmware upto date?

bridadan, 2017-01-17:
Also, this is for a release candidate for 5.1.0, which is quite old! Perhaps we should close this issue and reopen it with a test run from the latest release?

bulislaw, 2017-01-18:
Yeah, I'll close it and we can create a new issue if we see some issues on this board.

closed 2017-01-18 by bulislaw